wulf7/iichid #1: [WCOM50C1] Touch end events lost

myfreeweb, 2019-09-03:
hi, thanks for the work!

The Google Pixelbook has a Wacom touchscreen that supports both fingers and a pen:

<details>
<summary>Parsed Descriptor</summary>

```
0x05, 0x0D,        // Usage Page (Digitizer)
0x09, 0x04,        // Usage (Touch Screen)
0xA1, 0x01,        // Collection (Application)
0x85, 0x0C,        //   Report ID (12)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x15, 0x00,        //   Logical Minimum (0)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x54,        //   Usage (0x54)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0D,        //   Usage Page (Digitizer)
0x09, 0x22,        //   Usage (Finger)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x42,        //     Usage (Tip Switch)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x47,        //     Usage (0x47)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x05,        //     Report Count (5)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x10,        //     Report Size (16)
0x09, 0x51,        //     Usage (0x51)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x35, 0x00,        //     Physical Minimum (0)
0x46, 0xFF, 0x00,  //     Physical Maximum (255)
0x65, 0x11,        //     Unit (System: SI Linear, Length: Centimeter)
0x55, 0x0F,        //     Unit Exponent (-1)
0x75, 0x08,        //     Report Size (8)
0x09, 0x48,        //     Usage (0x48)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x49,        //     Usage (0x49)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x75, 0x10,        //     Report Size (16)
0x95, 0x01,        //     Report Count (1)
0x55, 0x0E,        //     Unit Exponent (-2)
0x65, 0x11,        //     Unit (System: SI Linear, Length: Centimeter)
0x09, 0x30,        //     Usage (X)
0x26, 0x80, 0x28,  //     Logical Maximum (10368)
0x35, 0x00,        //     Physical Minimum (0)
0x46, 0x20, 0x0A,  //     Physical Maximum (2592)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x46, 0xC0, 0x06,  //     Physical Maximum (1728)
0x09, 0x31,        //     Usage (Y)
0x26, 0x00, 0x1B,  //     Logical Maximum (6912)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x35, 0x00,        //     Physical Minimum (0)
0x45, 0x00,        //     Physical Maximum (0)
0x65, 0x00,        //     Unit (None)
0x55, 0x00,        //     Unit Exponent (0)
0x75, 0x08,        //     Report Size (8)
0x05, 0x0D,        //     Usage Page (Digitizer)
[... 514 lines cut ...]
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x02,        //     Report Count (2)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,NoWCOM50C1 Null Position)
0x95, 0x01,        //     Report Count (1)
0x75, 0x06,        //     Report Size (6)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x75, 0x10,        //     Report Size (16)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection

// 1289 bytes
```

</details>

The finger part works, but touches often get stuck without an end event..

(also there's a touchpad on this laptop but the ig4 is not seeing anything on its bus, I'm investigating that)

myfreeweb, 2019-09-05:
hm, USB `wmt` has the same behavior. the events look fine. i remember wmt working. The >2 finger thing is probably a problem up the stack (wlroots/wayfire)

wulf7, 2019-09-11:
Are touches stucks in interrupt mode?

myfreeweb, 2019-09-11:
I am using interrupt mode, but actually everything seems fine on the evdev level. In sway, I can press and release 4 fingers in gtk3-demo correctly, so it's a wayfire bug.

// I'd like to add Wacom pen support… do you have any ideas for allowing multiple drivers to attach for multiple usages under an `iichid`?

closed 2019-09-11 by myfreeweb

wulf7, 2019-09-12:
> // I'd like to add Wacom pen support… do you have any ideas for allowing multiple drivers to attach for multiple usages under an iichid?
I have some plans to implement HIDbus.
I am going to start from simple proxy driver that will allow to use unified HID drivers for both I2C and USB, and than extend it to support multiple childs

myfreeweb, 2019-09-12:
nice, I'll wait for that then