wulf7/iichid #7: Userspace HID access (/dev/uhidN)

myfreeweb, 2019-11-06:
Was wondering why the U2F mode on Yubikeys isn't working on my laptop (and only the laptop)…

```
ugen0.4: <Yubico Yubikey 4 OTP+U2F+CCID> at usbus0
ukbd0 on uhub0
ukbd0: <Yubico Yubikey 4 OTP+U2F+CCID, class 0/0, rev 2.00/4.35, addr 3> on usbus0
kbd2 at ukbd0
usbhid0 on uhub0
usbhid0: <Yubico Yubikey 4 OTP+U2F+CCID, class 0/0, rev 2.00/4.35, addr 3> on usbus0
hidbus2: <HID bus> on usbhid0
```

oops, we don't have the driver that exposes the userspace `/dev/uhidN` devices here

neelchauhan, 2019-11-06:
It could be that the Yubikey could use the same device ID as the USB Precision Touchpad, and `iichid` thinks the YubiKey is a touchpad.

myfreeweb, 2019-11-06:
`usbhid` from here matches on `{USB_IFACE_CLASS(UICLASS_HID)}`, it does not check for any touchpad device IDs, the only device ID it knows is the xbox 360 gamepad which is HID but does not have HID class

wulf7, 2019-11-06:
`uhid` - compatible child driver is planned but do not have great priority.
`usbhid` is adaptor to connect hidbus to USB subsystem. It does not detect any touchpads as it is up to the `hmt` driver to do that. It only detects generic USB HID device presence

cederom, 2020-02-21:
Hello world :-)

I also use YubiKey4NFC and YubiKey5NFC. They both expose USB HID for Push-Button authentication. These are very versatile devices that can have various functions enabled or disabled.

They DO work well as GPG/SSH agent key storage.

I did NOT manage to use it for web browser (firefox) U2F authentication even though they do work as usb keyboard and outputs alpha-numerical tokens after that magic shiny button is pressed.

What I usually use to work with them on FreeBSD:
```
gnupg-2.2.19                   Complete and free PGP implementation
yubico-piv-tool-1.7.0          Yubico PIV tool
yubikey-manager-qt-1.0.1_2     Cross-platform application for configuring any YubiKey
yubikey-personalization-gui-3.1.25_4 Graphical YubiKey personalization tool
yubioath-desktop-4.3.4_2       GUI for displaying OATH codes with a Yubikey
```

YubiKey4NFC:
```
ugen0.6: <Yubico Yubikey NEO OTP+CCID> at usbus0, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (30mA)

  bLength = 0x0012
  bDescriptorType = 0x0001
  bcdUSB = 0x0200
  bDeviceClass = 0x0000  <Probed by interface class>
  bDeviceSubClass = 0x0000
  bDeviceProtocol = 0x0000
  bMaxPacketSize0 = 0x0040
  idVendor = 0x1050
  idProduct = 0x0111
  bcdDevice = 0x0349
  iManufacturer = 0x0001  <Yubico>
  iProduct = 0x0002  <Yubikey NEO OTP+CCID>
  iSerialNumber = 0x0000  <no string>
  bNumConfigurations = 0x0001
```

YubiKey5NFC:
```
ugen0.6: <Yubico YubiKey CCID> at usbus0, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (30mA)

  bLength = 0x0012
  bDescriptorType = 0x0001
  bcdUSB = 0x0200
  bDeviceClass = 0x0000  <Probed by interface class>
  bDeviceSubClass = 0x0000
  bDeviceProtocol = 0x0000
  bMaxPacketSize0 = 0x0040
  idVendor = 0x1050
  idProduct = 0x0404
  bcdDevice = 0x0512
  iManufacturer = 0x0001  <Yubico>
  iProduct = 0x0002  <YubiKey CCID>
  iSerialNumber = 0x0000  <no string>
  bNumConfigurations = 0x0001
```

myfreeweb, 2020-02-21:
@cederom without the usb driver from here (and I think it's been disabled by default for now) it should work fine. Make sure you have permission to access the uhid devices (there's a package with devd rules for that)

myfreeweb, 2020-03-25:
`hidraw.ko` is there now, and with #27 (`uhid` alias) u2f works for me, so closing.

closed 2020-03-25 by myfreeweb