wulf7/iichid #8: hconf does not find input_mode

myfreeweb, 2019-11-15:
The new hconf code cannot find the input mode feature on the Pixelbook touchpad…

<details><summary>Parsed descriptor</summary>

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x02,        //     Usage Maximum (0x02)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //     Report Count (6)
0x81, 0x03,        //     Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x02,        //     Report Count (2)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
0x05, 0x0D,        // Usage Page (Digitizer)
0x09, 0x05,        // Usage (Touch Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x05, 0x09,        //   Usage Page (Button)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x01,        //   Usage Maximum (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x01,        //   Report Count (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0D,        //   Usage Page (Digitizer)
0x09, 0x54,        //   Usage (0x54)
0x25, 0x05,        //   Logical Maximum (5)
0x75, 0x07,        //   Report Size (7)
0x95, 0x01,        //   Report Count (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0D,        //   Usage Page (Digitizer)
0x09, 0x22,        //   Usage (Finger)
0xA1, 0x02,        //   Collection (Logical)
0x09, 0x47,        //     Usage (0x47)
0x09, 0x42,        //     Usage (Tip Switch)
0x09, 0x32,        //     Usage (In Range)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x75, 0x01,        //     Report Size (1)
0x95, 0x03,        //     Report Count (3)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x51,        //     Usage (0x51)
0x25, 0x1F,        //     Logical Maximum (31)
0x75, 0x05,        //     Report Size (5)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x55, 0x0E,        //     Unit Exponent (-2)
0x65, 0x11,        //     Unit (System: SI Linear, Length: Centimeter)
0x35, 0x00,        //     Physical Minimum (0)
[... 321 lines cut ...]
					conf_coll = false;
				break;
			case hid_feature:
				if (conf_coll && hi.usage == HID_USAGE2(HUP_DIGITIZERS, HUD_INPUT_MODE)) {
					sc->input_mode_rid = hi.report_ID;
					if (sc == NULL) {
						device_printf(dev,"failed to locate old style\n");
						hid_end_parse(hd);
						break;
					}
					sc->input_mode_loc = hi.loc;
					sc->input_mode_rlen = hid_report_size_1(d_ptr, d_len,
					    hid_feature, sc->input_mode_rid);
				}
			default: break;
			}
		}
		hid_end_parse(hd);
	}

	return (0);
}
```

</details>

wulf7, 2019-11-16:
try to replace HUD_CONFIG with HUD_INPUT_MODE in hid_tlc_locate parameters

myfreeweb, 2019-11-16:
That doesn't help.

Again, the body of the `HID_TLC_FOREACH_ITEM` loop does not get executed, so it doesn't get to the point where it compares features.. maybe the magic macro is broken for the last TLC or something? I'll try debugging with a manual loop that does similar things

myfreeweb, 2019-11-16:
Oh. There was also a problem with the macro. `iter += (… ? 1 : 0) == tidx` should be `(iter += (… ? 1 : 0)) == tidx`! (`INPUT_MODE` was also required)

closed 2019-11-16 by myfreeweb

wulf7, 2019-11-16:
Parenthesis committed. Thanks!
Strange enough, the buggy code worked well with second TLC while I was testing it with my touchscreen's firmware downloader TLC