wulf7/iichid #10: Problems with ELAN1200

arrowd, 2019-11-27:
Here is result of trying this driver on my laptop:

```
iichid0: HID command I2C_HID_CMD_DESCR at 0x1
iichid0: HID descriptor: 1e 00 00 01 64 01 02 00 03 00 10 00 04 00 00 00 05 00 06 00 f3 04 45 30 11 00 00 00 00 00
iichid0 at addr 0x15 irq 117 on iicbus0
iichid0:   ACPI Hardware ID  : ELAN1200
iichid0:   IICbus addr       : 0x15
iichid0:   HID descriptor reg: 0x01
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: HID command I2C_HID_CMD_RESET
iichid0: HID command I2C_HID_REPORT_DESCR at 0x2 with size 356
iichid0: HID report descriptor: 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 75 01 95 02 81 02 95 06 81 03 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 05 0c 0a 38 02 95 01 81 06 75 08 95 03 81 03 c0 c0 05 0d 09 05 a1 01 85 04 09 22 a1 02 15 00 25 01 09 47 09 42 95 02 75 01 81 02 75 01 95 02 81 03 95 01 75 04 25 0f 09 51 81 02 05 01 15 00 26 80 0c 75 10 55 0e 65 13 09 30 35 00 46 90 01 95 01 81 02 46 13 01 26 96 08 26 96 08 09 31 81 02 05 0d 15 00 25 64 95 03 c0 55 0c 66 01 10 47 ff ff 00 00 27 ff ff 00 00 75 10 95 01 09 56 81 02 09 54 25 7f 95 01 75 08 81 02 05 09 09 01 25 01 75 01 95 01 81 02 95 07 81 03 09 c5 75 08 95 04 81 03 05 0d 85 02 09 55 09 59 75 04 95 02 25 0f b1 02 85 07 09 60 75 01 95 01 15 00 25 01 b1 02 95 0f b1 03 06 00 ff 06 00 ff 85 06 09 c5 15 00 26 ff 00 75 08 96 00 01 b1 02 85 0d 09 c5 15 00 26 ff 00 75 08 95 04 b1 02 85 0c 09 c6 96 8a 02 75 08 b1 02 85 0b 09 c7 95 80 75 08 b1 02 c0 05 0d 09 0e a1 01 85 03 09 22 a1 00 09 52 15 00 25 0a 75 08 95 02 b1 02 c0 09 22 a1 00 85 05 09 57 09 58 15 00 75 01 95 02 25 03 b1 02 95 0e b1 03 c0 c0
iichid0: allocated irq at 0xfffff80002670000 and rid 0
iichid0: successfully setup interrupt
iichid0: HID command I2C_HID_CMD_SET_POWER(1)
acpi_iichid0: <HID over I2C (ACPI)> on acpi0
hidbus0: <HID bus> on iichid0
hms0 on hidbus0
hms0: 2 buttons and [XYZT] coordinates ID=1
hmt0 on hidbus0
iichid0: HID command I2C_HID_CMD_GET_REPORT 2 (type 3, len 2)
iichid0: response: 04 00 02 05
iichid0: HID command I2C_HID_CMD_GET_REPORT 6 (type 3, len 257)
iichid0: response: 03 01 06 fc 28 fe 84 40 cb 9a 87 0d be 57 3c b6 70 09 88 07 97 2d 2b e3 38 34 b6 6c ed b0 f7 e5 9c f6 c2 2e 84 1b e8 b4 51 78 43 1f 28 4b 7c 2d 53 af fc 47 70 1b 59 6f 74 43 c4 f3 47 18 53 1a a2 a1 71 c7 95 0e 31 55 21 d3 b5 1e e9 0c ba ec b8 89 19 3e b3 af 75 81 9d 53 b9 41 57 f4 6d 39 25 29 7c 87 d9 b4 98 45 7d a7 26 9c 65 3b 85 68 89 d7 3b bd ff 14 67 f2 2b f0 2a 41 54 f0 fd 2c 66 7c f8 c0 8f 33 13 03 f1 d3 c1 0b 89 d9 1b 62 cd 51 b7 80 b8 af 3a 10 c1 8a 5b e8 8a 56 f0 8c aa fa 35 e9 42 c4 d8 55 c3 38 cc 2b 53 5c 69 52 d5 c8 73 02 38 7c 73 b6 41 e7 ff 05 d8 2b 79 9a e2 34 60 8f a3 32 1f 09 78 62 bc 80 e3 0f bd 65 20 08 13 c1 e2 ee 53 2d 86 7e a7 5a c5 d3 7d 98 be 31 48 1f fb da af a2 a8 6a 89 d6 bf f2 d3 32 2a 9a e4 cf 17 b7 b8 f4 e1 33 08 24 8b c4 43 a5 e5 24 c2
hconf0 on hidbus0
iichid0: HID command I2C_HID_CMD_GET_REPORT 3 (type 3, len 3)
iichid0: response: 05 00 03 00 00
iichid0: HID command I2C_HID_CMD_SET_REPORT 3 (type 3, len 3): 03 03 00
hmt0: Multitouch touchpad with 1 button, click-pad
hmt0: 5 contacts with [C] properties. Report range [0:0] - [3200:2198]
```

Looks promising. However, touching the touchpad results in messages:

```
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
```

Will be happy to provide any additional info or debug something.

wulf7, 2019-11-27:
You should not get any events until touchpad's cdev is opened with Xorg/libinput or some evdev-debugging tool like evemu-record. So this looks like unfiltered noise

arrowd, 2019-11-27:
Do I need some special setup to make Xorg pick it up?

wulf7, 2019-11-27:
It is better to start with evemu-record (from devel/evemu).
Run it and select one of ELAN1200 devices, matching touchpad. It has EV_ABS events rather than EV_REL. Than touch surface and check if any events appeared on screen.

arrowd, 2019-11-27:
For some reason I have 2 ELAN1200 devices, but the second one seems to work - I see a stream of events when touching the touchpad.

wulf7, 2019-11-27:
> For some reason I have 2 ELAN1200 devices

That is normal for touchpads. First device emulates relative mouse and exists for compatibility mostly. And second one is fully functional absolute multitouch touchpad. hconf0 device makes one of them active.
> I see a stream of events when touching the touchpad.

Most probably your touchpad works with iichid.ko
> Do I need some special setup to make Xorg pick it up?

The simplest way is setup some evdev-awared Xorg input autoconfigurator like https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=196678
It is possible to configure Xorg manually, but you should google for HOW TOs yourself. I forgot how to do that.

arrowd, 2019-11-27:
I'll better wait for all this work to be upstreamed. This can be closed, I think.

Thanks for all the pointers and explanations!

closed 2019-11-27 by arrowd