wulf7/iichid #12: buiding with 13-CURRENT

omatsuda, 2020-01-24:
Thank you very much for this nice software.

I recently bought a new laptop Fujitsu WU2/D2 which has no PS/2 interface for its built-in touchpad.  In addition, it has a relatively new Cannon Point-LP Serial IO I2C Controller which has not been supported yet by ig4 in 12.1-RELEASE.  So I need to use iichid with 13-CURRENT.

kbd.c of iichid uses genkbd_diag() and genkbd_get_fkeystr() which used to be defined in /usr/src/sys/dev/kdb/kbdreg.h.  However, at some moment (probably Dec. 16, 2019), these functions become static in 13-CURRENT and the definitions in kbdreg.h were removed.

During the build of iichid with 13-CURRENT (I used the snapshot FreeBSD-13.0-CURRENT-amd64-20200116-r356767), the compiler suggested the usage of kbdd_diag an dkbdd_get_fkeystr instead.  (I don't know how the compiler knew this...)

Anyway the patch below allows the build successful.  I haven't checked whether this change affects the functionality of the program, though.

-------------------
*** hkbd.c.orig	Tue Dec 17 07:29:04 2019
--- hkbd.c	Sat Jan 18 01:43:57 2020
***************
*** 1117,1123 ****
  	sc->sc_flags |= HKBD_FLAG_ATTACHED;
  
  	if (bootverbose) {
! 		genkbd_diag(kbd, bootverbose);
  	}
  
  	/* start the keyboard */
--- 1117,1123 ----
  	sc->sc_flags |= HKBD_FLAG_ATTACHED;
  
  	if (bootverbose) {
! 		kbdd_diag(kbd, bootverbose);
  	}
  
  	/* start the keyboard */
***************
*** 2077,2085 ****
  	.clear_state = &hkbd_clear_state,
  	.get_state = &hkbd_get_state,
  	.set_state = &hkbd_set_state,
! 	.get_fkeystr = &genkbd_get_fkeystr,
  	.poll = &hkbd_poll,
! 	.diag = &genkbd_diag,
  };
  
  KEYBOARD_DRIVER(hkbd, hkbdsw, hkbd_configure);
--- 2077,2085 ----
  	.clear_state = &hkbd_clear_state,
  	.get_state = &hkbd_get_state,
  	.set_state = &hkbd_set_state,
! 	.get_fkeystr = &kbdd_get_fkeystr,
  	.poll = &hkbd_poll,
! 	.diag = &kbdd_diag,
  };
  
  KEYBOARD_DRIVER(hkbd, hkbdsw, hkbd_configure);

omatsuda, 2020-01-24:
Sorry.  I should not have included the patch in the text.  Please see the one attached to this.

[hkbd.diff.txt](https://github.com/wulf7/iichid/files/4106781/hkbd.diff.txt)

skaarj0, 2020-02-02:
One question:  does this build expose the peripherals to the user in /dev?  It does not seem to work in FreeBSD 12.1-stable - there's no /dev/hms* or any other new peripheral name in /dev. Please excuse my non-specific terms, as I am just starting to learn about creating FreeBSD drivers.

wulf7, 2020-02-03:
> One question: does this 13-CURRENT build expose the peripherals to the user in /dev? It does not seem to work in 12.1-STABLE - there's no /dev/hms* or any other new peripheral name in /dev. Please excuse my non-specific terms, as I am just starting to learn about creating FreeBSD drivers.

Yes, it does. Look for /dev/input/event*

wulf7, 2020-02-03:
Thank you for the patch, @omatsuda. I applied latest upstream ukbd(4) changes as 5d3a4a2a so hkbd.c is buildable on recent CURRENT again.

omatsuda, 2020-02-04:
I confirmed the latest version (committed on Jan. 29) is buildable and works fine also on 12.1-STABLE (r357392 in my case).  

Note that Cannon Point-LP Serial IO I2C Controller has already been supported in 12.1-STABLE, so  I moved down to 12.1-STABLE.

closed 2020-02-06 by omatsuda