wulf7/iichid #13: touchpad buttons

omatsuda, 2020-02-04:
Thank you very much again for the nice program.

After successful built in 13-CURRENT and 12-STABLE), the touchpad on my Fujitsu WU2/D2 laptop mostly works, but there remains an annoying problem.

The physical left button is recognized as the right button, and the physical right button is recognized as the middle button. Though this can be rectified by the button mapping fascilities, there is another problem: the middle button emulation in libinput doesn't work.  I guess this emulation assumes the physical buttons were naturally recognized as the left and right buttons.

The button recognition seems to be done in hmt.c around HMT_FOREACH_BUTTONS macro.  In my case, the attached patch 
[hmt.diff.txt](https://github.com/wulf7/iichid/files/4154579/hmt.diff.txt)
solved all the problems: the buttons are recognized correctly, and the middle button emulation works fine.

The attached patch also tells us a bit how the driver recognizes the buttons.  Note BTN_MOUSE et al, are defined in /usr/include/dev/evdev/input-event-codes.h as

BTN_MOUSE 0x110
BTN_LEFT 0x110
BTN_RIGHT 0x111
BTN_MIDDLE 0x112

What the patch does is to shift
  BTN_RIGHT and BTN_MIDDLE
to
  BTN_LEFT and BTN_RIGHT

wulf7, 2020-02-05:
Thank you for report and patch @omatsuda -san

As some other tested devices (clickpads) does button to event mapping properly for me, I think the bug is lying elsewhere.

Could you provide HID and report descriptors from your I2C device?
Just copy-paste terminal output of `kldload ./iichid.ko` here.

omatsuda, 2020-02-05:
Thank you for your reply.

Here is the terminal output of `kldload ./iichid.ko`
[iichid.orig.log.txt](https://github.com/wulf7/iichid/files/4160182/iichid.orig.log.txt)
iichid.ko is compiled from the original source files.

wulf7, 2020-02-06:
> Here is the terminal output of `kldload ./iichid.ko`

Hmm... Report descriptor really defines touchpad buttons as 'button 2' (right) and 'button 3' (middle). That looks wrong.
Let's try to workaround this with renumbering of buttons starting from 'button 1'

Please try latest iichid revision.

omatsuda, 2020-02-07:
Thank you very much!
The latest iichid on Feb. 7, 2020 worked perfectly on my Fujitsu WU2/D2 laptop with Synaptics touchpad SYNP1F13 on I2C.
Here is the terminal output of `kldload ./iichid.ko`
[iichid.200207.log.txt](https://github.com/wulf7/iichid/files/4168835/iichid.200207.log.txt)

ara-ler, 2020-02-07:
Thank for the the driver! Works great in 13.0 on Dell Precision 7540.
Regarding the buttons: my left and right buttons work as expected, but not the middle one. I'm ready to test and provide more info.

Is this driver being considered to be included in 13.0? What's the most appropriate place for such discussions as well as for other general comments and experience sharing?

wulf7, 2020-02-07:
> Regarding the buttons: my left and right buttons work as expected, but not the middle one. I'm ready to test and provide more info.

Open new issue and post HID and report descriptors from your I2C device with copy-pasting of terminal output of `kldload ./iichid.ko` command.
> Is this driver being considered to be included in 13.0?

Yes. It is a goal,
> What's the most appropriate place for such discussions as well as for other general comments and experience sharing?

I don't know. There were some discussions at previous I2C HID driver review: https://reviews.freebsd.org/D16698

wulf7, 2020-02-07:
> The latest iichid on Feb. 7, 2020 worked perfectly on my Fujitsu WU2/D2 laptop with Synaptics touchpad SYNP1F13 on I2C.

It is great to hear that. Thanks!
> Here is the terminal output of `kldload ./iichid.ko`
> [iichid.200207.log.txt](https://github.com/wulf7/iichid/files/4168835/iichid.200207.log.txt)
> This is exactly the same as iichid.orig.log.txt which I uploaded before.

That is expected.

ara-ler, 2020-02-07:
Here is the corresponding part from dmesg on Precision 7540 after loading iichid.ko:
[iichid-dell7540.txt](https://github.com/wulf7/iichid/files/4172809/iichid-dell7540.txt)

ara-ler, 2020-02-07:
...forgot to mention that Precision 7540 has both trackpoint and touchpad, so I got 2 corresponding device nodes:
/dev/input/event3
/dev/input/event4

wulf7, 2020-02-10:
> ...forgot to mention that Precision 7540 has both trackpoint and touchpad, so I got 2 corresponding device nodes:
> /dev/input/event3
> /dev/input/event4

Most probably your trackpoint is attached to PS/2 controller not to I2C one and irrelevant for us.
/dev/input/event3 should be a I2C faked mouse and is mostly required to show mouse pointer on Win 8.1. It should be silent all the time except you enable it with  `# sysctl dev.hconf.0.input_mode=0` command.

> Here is the corresponding part from dmesg on Precision 7540 after loading iichid.ko:

Unfortunatelly, I can't find any traces of 3-rd button in report descriptor. Could you dump raw I2C HID reports while pressing and releasing 3-rd button? Type `# sysctl hw.iichid.debug=5` to enable dumping of input reports to system console.

P.S. both sysctls require latest iichid.

ara-ler, 2020-02-10:
Attached is the log with debug=5, each button was pressed once.
The middle button's codes are totally different from the other two's.
[iichid-buttons123.txt](https://github.com/wulf7/iichid/files/4182365/iichid-buttons123.txt)

wulf7, 2020-02-10:
> Attached is the log with debug=5, each button was pressed once.
> The middle button's codes are totally different from the other two's.
> [iichid-buttons123.txt](https://github.com/wulf7/iichid/files/4182365/iichid-buttons123.txt)

The middle button is reported as faked mouse's middle button rather than touchpad's button.
Try to attach evemu-record (from devel/evemu) to first DELL0926 device (should be /dev/input/event3) and see if middle button presses are reported here.

ara-ler, 2020-02-10:
That's true! The middle button events go to /dev/input/event3. Good to know, thanks!
Not a big deal, 3-finger emulation works great.
Also, with the latest iichid.ko the left button's events do not go /dev/input/event4, i.e. it doesn't work in X.

wulf7, 2020-02-10:
> Also, with the latest iichid.ko the left button's events do not go /dev/input/event4, i.e. it doesn't work in X.
Try updated iichid

ara-ler, 2020-02-10:
It's working now, thanks!

closed 2020-02-21 by omatsuda