wulf7/iichid #14: Use x.y.z version scheme and add git tags

koobs, 2020-02-12:
I'm about to land a sysutils/iichid port via [FreeBSD Review 23615](https://reviews.freebsd.org/D23615) and it would be handy going forward to have an `x.y.z` version scheme set for the code, with git tags added so that the port can fetch using those tags instead of git commit hash

The expectation with the port, is that even if/when iichid lands in base, that the port will continue to serve as an easy test method for in development iichid code, in advance of updates landing in base, to facilitate early feedback from users for new features and bugfixes alike.

We can start with 0.0.1 for the current code (or another x.y.z version if you think its more suitable)

wulf7, 2020-02-12:
Latest commit is tagged as 0.0.1

koobs, 2020-02-14:
Fetched using 0.0.1, thank you!

closed 2020-02-14 by koobs