wulf7/iichid #16: Update README

koobs, 2020-02-14:
Now that the sysutils/iichid port has landed, update README to include at least:

* What kinds of devices *might* be expected to work (device classes, examples etc)
* Basic instructions for how users can check/verify if something is supported/works. Maybe something from   https://larastechlab.blogspot.com/2020/01/i2c-touchpads-on-freebsd-130-current.html might be handy
* Notes on what information/feedback you want that would be helpful, and in what form

wulf7, 2020-03-01:
Done in f019672 and 76a9a01

koobs, 2020-03-02:
Thanks @wulf7!

If it would be handy for people to include additional system/hardware information (like pciconf, usbconfig or var/run/dmesg.boot) when reporting bugs, id would be great to include those commands in the README

closed 2020-03-02 by koobs