wulf7/iichid #17: KLD iichid.ko: depends on kernel - not available or version mismatch

cederom, 2020-02-21:
Hello @wulf7 and big thank you for creating this generic HID layer for FreeBSD !! :-)

I have problem loading `iichid 0.0.1` module on a `12.1-RELEASE-p1` when built with `stable/12` sources..

```
root@0xCFMX4:/usr/src # svn info
Path: .
Working Copy Root Path: /usr/src
URL: https://svn.freebsd.org/base/stable/12
Relative URL: ^/stable/12
Repository Root: https://svn.freebsd.org/base
Repository UUID: ccf9f872-aa2e-dd11-9fc8-001c23d0bc1f
Revision: 358209
Node Kind: directory
Schedule: normal
Last Changed Author: pfg
Last Changed Rev: 358209
Last Changed Date: 2020-02-21 05:48:25 +0100 (Fri, 21 Feb 2020)

root@0xCFMX4:/usr/src # uname -a
FreeBSD 0xCFMX4 12.1-RELEASE-p1 FreeBSD 12.1-RELEASE-p1 GENERIC  amd64
```

wulf7, 2020-02-21:
Please, rebuild your kernel. You MUST compile both kernel and iichid.ko from the same sources.

cederom, 2020-02-21:
I have reverted `/usr/src` to `12.1-RELEASE` and module builds and loads fine :-)

**Great Great thank you @wulf7 for this project!** Not only I hope to use it for my laptop (PS/2 Synaptics) Trackpad gestures but also advanced DualShock2 (PS4) controller (touchpad + IMU + hopefully Bluetooth as well) :-)

```
 # svn info
Path: .
Working Copy Root Path: /usr/src
URL: https://svn.freebsd.org/base/release/12.1.0
Relative URL: ^/release/12.1.0
Repository Root: https://svn.freebsd.org/base
Repository UUID: ccf9f872-aa2e-dd11-9fc8-001c23d0bc1f
Revision: 358209
Node Kind: directory
Schedule: normal
Last Changed Author: gjb
Last Changed Rev: 354337
Last Changed Date: 2019-11-04 19:34:29 +0100 (Mon, 04 Nov 2019)

# uname -a
FreeBSD 0xCFMX4 12.1-RELEASE-p1 FreeBSD 12.1-RELEASE-p1 GENERIC  amd64

root@0xCFMX4:/usr/ports/sysutils/iichid # make deinstall
root@0xCFMX4:/usr/ports/sysutils/iichid # make clean
root@0xCFMX4:/usr/ports/sysutils/iichid # make
root@0xCFMX4:/usr/ports/sysutils/iichid # make install

root@0xCFMX4:/usr/ports/sysutils/iichid # kldunload ig4
root@0xCFMX4:/usr/ports/sysutils/iichid # kldload ig4
root@0xCFMX4:/usr/ports/sysutils/iichid # kldload iichid
root@0xCFMX4:/usr/ports/sysutils/iichid # kldstat
Id Refs Address                Size Name
 1  129 0xffffffff80200000  2448ef8 kernel
 2    1 0xffffffff82649000   3a99a8 zfs.ko
 3    2 0xffffffff829f3000     a5b8 opensolaris.ko
 4    1 0xffffffff829fe000      ec0 green_saver.ko
 5    1 0xffffffff829ff000     e1d8 aesni.ko
 6    1 0xffffffff82a0e000    994a8 linux64.ko
 7    5 0xffffffff82aa8000     9698 linux_common.ko
 8    1 0xffffffff82ab2000     3170 splash_bmp.ko
 9    1 0xffffffff82ab6000    262b0 fuse.ko
10    1 0xffffffff82add000     bbe8 autofs.ko
11    1 0xffffffff82ae9000     c408 cuse.ko
12    1 0xffffffff82af6000    1e8a0 geom_eli.ko
13    1 0xffffffff82b16000    b7bd8 linux.ko
14    1 0xffffffff82c21000     1aa0 fdescfs.ko
15    1 0xffffffff82c23000   12c8b0 i915kms.ko
16    1 0xffffffff82d50000    763b0 drm.ko
17    4 0xffffffff82dc7000    10eb0 linuxkpi.ko
18    3 0xffffffff82dd8000    12f30 linuxkpi_gplv2.ko
19    2 0xffffffff82deb000      6d0 debugfs.ko
20    1 0xffffffff82dec000    fb241 iwm7265Dfw.ko
21    1 0xffffffff82ee8000    15d20 if_iwm.ko
22    1 0xffffffff82efe000     3f30 vboxnetadp.ko
23    2 0xffffffff82f02000    531c0 vboxdrv.ko
24    1 0xffffffff82f56000     2ce0 vboxnetflt.ko
25    2 0xffffffff82f59000     9e30 netgraph.ko
26    1 0xffffffff82f63000     1710 ng_ether.ko
27    1 0xffffffff82f65000     18a0 uhid.ko
28    1 0xffffffff82f67000     1aa0 wmt.ko
29    1 0xffffffff82f69000     4c80 u3g.ko
30    1 0xffffffff82f6e000     43c0 ucom.ko
31    1 0xffffffff82f73000    258f8 ipfw.ko
32    1 0xffffffff82f99000     494c linprocfs.ko
33    1 0xffffffff82f9e000     1eae linsysfs.ko
34    1 0xffffffff82fa0000     88d8 tmpfs.ko
40    1 0xffffffff82fa9000     35a8 ig4.ko
41    3 0xffffffff82fad000     1fa0 iicbus.ko
42    1 0xffffffff82faf000     9250 iichid.ko
```

closed 2020-02-21 by cederom