wulf7/iichid #18: Handbook / WIKI / manpage

cederom, 2020-02-21:
It would be really really nice to spread the news already about this project on Touchpad WIKI and FreeBSD Hanbook :-)

man page would be also nice to have when project gets out from experimental stage :-)

I got here "by accident" while this is very important milestone in FreeBSD HID implementation, thank you!! :-)

koobs, 2020-02-26:
Manual pages and Handbook entries will come when it hits base. Until then any additional info related to playing with or using this module could be put in the FreeBSD wiki

koobs, 2020-02-26:
See also: #16 for README updating

@wulf we can now close this general support issue/question :)

cederom, 2020-02-26:
Thanks @koobs and @wulf7 I am aware of TODO in README.. but it's not really about README.

People are unaware of this project and it would be really nice to at lease mention it on the Synaptics Touchpad WIKI [1] where they look in the first place when problems with trackpad occurs. Unfortunately I have no access to edit it.

Handbook and MAN after a release seems reasonable - so far this GitHub project page is a nice documentation and support site :-)

As you please - you can close it now, or keep it open as TODO reminder and close it when the job is done :-)

[1] https://wiki.freebsd.org/SynapticsTouchpad

koobs, 2020-03-02:
Added sysutils/iichid port description and link to Wiki page in [SynapticsTouchpad: Revision22](https://wiki.freebsd.org/action/diff/SynapticsTouchpad?action=diff&rev1=21&rev2=22)

wulf7, 2020-03-02:
Wiki page is mostly outdated after recent 1.20.7 Xorg update. Setting kern.evdev.rcpt_mask sysctl to 6 or 12 effectively disables all the synaptics sysctls except hw.psm.synaptics.min_pressure

koobs, 2020-03-02:
Ping @zeising re updating wiki page to update/remove stale bits post xorg-server 1.20.x update