wulf7/iichid #20: Cursor/pointer motion continues "on its own"

perhedeland, 2020-02-21:
Hi, this driver has made my dead touchpad (ELAN0608 on Lenovo ideapad 320) come to life, thanks a *lot* for that! There is a weird issue though - if I move the cursor/pointer and let go of the touchpad without first stopping the motion, the cursor continues to move on its own at the same speed until it hits the closest edge of the screen (also continues along that edge until it reaches a corner). It can 
pretty easily be avoided by stopping the motion before letting go of the touchpad, but it does seem like it shouldn't happen...

I have no idea if this is something that can be addressed in this driver, but if there are any patches or the like that I can try out, I'll be happy to do so. I'm using the FreeBSD port version 0.0.1 on 12.1-RELEASE.

wulf7, 2020-02-21:
Hi,

Try to upgrade your kernel to latest stable and rebuild iichid than. If you are lucky enough this will enable interrupt mode in the driver and this weird motion will go away.
It will not help if your touchpad uses GPIO interrupts which are not supported by FreeBSD yet but it is worth trying.

perhedeland, 2020-02-22:
Thanks - unfortunately, with the 12-STABLE I tried (r358121), the touchpad is dead again (yes, I remembered to rebuild iichid:-). Maybe I should try CURRENT?

wulf7, 2020-02-22:
> Maybe I should try CURRENT?

IIRC all relevant commits were MFC-ed before new year, so no need.

Could you place console output after `sudo kldload iichid.ko` here?

perhedeland, 2020-02-22:
Well, from dmesg, but it seems to be identical to what is printed on the console (also seems identical to what I got on 12.1-RELEASE except for the address after "allocated irq at"):

iichid0: HID command I2C_HID_CMD_DESCR at 0x1
iichid0: HID descriptor: 1e 00 00 01 7c 00 02 00 03 00 22 00 04 00 00 00 05 00 06 00 f3 04 00 04 01 00 00 00 00 00
iichid0 at addr 0x15 irq 87 on iicbus0
iichid0:   ACPI Hardware ID  : ELAN0608
iichid0:   IICbus addr       : 0x15
iichid0:   HID descriptor reg: 0x01
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: HID command I2C_HID_CMD_RESET
iichid0: HID command I2C_HID_REPORT_DESCR at 0x2 with size 124
iichid0: HID report descriptor: 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 75 01 95 02 81 02 95 06 81 03 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 05 0c 0a 38 02 95 01 81 06 75 08 95 03 81 03 c0 c0 06 00 ff 06 00 ff 09 01 a1 01 85 5d 09 01 95 1f 75 08 81 06 85 0d 09 c5 15 00 26 ff 00 75 08 95 04 b1 02 85 0c 09 c6 96 10 04 75 08 b1 02 85 0b 09 c7 95 42 75 08 b1 02 c0
hid_get_item: Number of items(1040) truncated to 1024
iichid0: allocated irq at 0xfffff80003c9b880 and rid 0
iichid0: successfully setup interrupt
iichid0: HID command I2C_HID_CMD_SET_POWER(1)
acpi_iichid0: <HID over I2C (ACPI)> on acpi0
hidbus0: <HID bus> on iichid0
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hms0 on hidbus0
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hms0: 2 buttons and [XYZT] coordinates ID=1
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024
hid_get_item: Number of items(1040) truncated to 1024

perhedeland, 2020-02-22:
Btw, it seems the STABLE kernel auto-loads ig4 and iicbus for me, while I previously had/needed

kld_list="/boot/modules/i915kms.ko vboxdrv ig4 iicbus iichid"

in rc.conf - I still have it, but I commented-out the whole line and rebooted in order to get the output in the previous comment - and noticed that ig4 and iicbus were already loaded despite of this.

perhedeland, 2020-02-22:
Sorry, I only now noticed that the touchpad was on /dev/input/event5 instead of /dev/input/event3 where it was before - updated the xorg.conf snippet accordingly, and the touchpad works again - and the "moving on its own" issue is gone! Thanks a lot again!

perhedeland, 2020-02-22:
Apropos that, I see that UDEV is enabled by default in the latest version of the xorg-server port (released 2 days ago:-) - I guess I should upgrade...

closed 2020-02-22 by perhedeland