wulf7/iichid #21: Two-finger scroll doesn't work with ELAN0608 touchpad

perhedeland, 2020-03-01:
Hi,
I think I'm abusing the "issue" functionality here, since I have no reason to think that this is a problem with the iichid module - but since it was what got my touchpad working *at all* (thanks again!), I thought you might have some idea about how to get the two-finger scroll to work too...

After you helped me out in issue #20, I have upgraded to the latest xorg-server port with UDEV enabled, and even gotten it to work:-) with the help of Michael Gmelin's excellent script - no more need to specify the exact /dev/input/eventN in an xorg.conf snippet, that part "just works".

But one argument for UDEV that has been mentioned is to get more touchpad functionality working, in particular the two-finger scroll - and it doesn't work for me. I know that the touchpad supports it, since it works in Windows (I have the laptop set up for dual-boot). I *think* Xorg.0.log has a significant clue, in that it decides that my touchpad is a mouse:-) -

```
-------------------------------------------------------------
[    29.512] (II) config/udev: Adding input device ELAN0608 (/dev/input/event3)
[    29.512] (**) ELAN0608: Applying InputClass "evdev pointer catchall"
[    29.512] (**) ELAN0608: Applying InputClass "libinput pointer catchall"
[    29.512] (II) Using input driver 'libinput' for 'ELAN0608'
[    29.512] (**) ELAN0608: always reports core events
[    29.512] (**) Option "Device" "/dev/input/event3"
[    29.512] (**) Option "_source" "server/udev"
[    29.512] (II) event3  - ELAN0608: is tagged by udev as: Mouse
[    29.512] (II) event3  - ELAN0608: device is a pointer
[    29.512] (II) event3  - ELAN0608: device removed
[    29.512] (**) Option "config_info" "udev:/dev/input/event3"
[    29.512] (II) XINPUT: Adding extended input device "ELAN0608" (type: MOUSE, id 9)
[    29.513] (**) Option "AccelerationScheme" "none"
[    29.513] (**) ELAN0608: (accel) selected scheme none/0
[    29.513] (**) ELAN0608: (accel) acceleration factor: 2.000
[    29.513] (**) ELAN0608: (accel) acceleration threshold: 4
[    29.513] (II) event3  - ELAN0608: is tagged by udev as: Mouse
[    29.513] (II) event3  - ELAN0608: device is a pointer
-------------------------------------------------------------
```
Do you have any idea? I can easily try out any source changes.

wulf7, 2020-03-01:
Unfortunately your touchpad uses Elantech proprietary protocol rather than HID for multitouch mode which is not supported in this driver.
It is possible to port Elantech driver from Linux which should support this device: https://elixir.bootlin.com/linux/v5.6-rc3/source/include/linux/input/elan-i2c-ids.h#L37 but I do not have opportunities to do that.

perhedeland, 2020-03-02:
On 2020-03-02 00:02, Vladimir Kondratyev wrote:
> Unfortunately your touchpad uses Elantech proprietary protocol rather than HID for multitouch mode which is not supported in this driver.

Ouch, too bad - I was hoping I could actually use the touchpad as a
full replacement for a mouse, but I find that I've become too
dependant on the mouse wheel...

> It is possible to port Elantech driver from Linux which should support this device: https://elixir.bootlin.com/linux/v5.6-rc3/source/include/linux/input/elan-i2c-ids.h#L37 but I do not have
> opportunities to do that.

Hm, while I have "tinkered" with both the Linux and the FreeBSD
kernels in the past, it was mostly "simple" stuff, no device drivers -
such a port is definitely beyond my *current* capabilities, but I'll
try to at least have a look... Thanks anyway!

--Per

wulf7, 2020-04-29:
There is not yet committed OpenBSD driver for Elan I2C touchpad https://marc.info/?l=openbsd-tech&m=158509428830926
Porting it to FreeBSD looks rather straightforward. One is to replace OpenBSD bus bits (probe, interrupt establishing and I2C transfers) with newbus bits taken from iichid.c and replace wscons interface with evdev.

perhedeland, 2020-04-29:
On 2020-04-29 15:11, Vladimir Kondratyev wrote:
> There is not yet committed OpenBSD driver for Elan I2C touchpad https://marc.info/?l=openbsd-tech&m=158509428830926
> Porting it to FreeBSD looks rather straightforward. One is to replace OpenBSD bus [...]

Thanks, sounds great! I can't say that I understand which bits need to
be changed and how, but I'll at least *try* to figure it out...

--Per

wulf7, 2020-11-29:
I added a port of "Linux Elan I2C touchpad" driver to iichid, so you can test it. I do not have a hardware to do that.

studikumbang, 2021-08-31:
FreeBSD 13.0-STABLE amd64

wulf7, 2021-09-02:
Elantech proprietary protocol support was not imported into 13.0 due to lack of testing.
You may create FreeBSD bugzilla issue and assign it to me to test patches as this version of iichid does not compile on 13.0-STABLE