wulf7/iichid #28: Dell 7490 touchpad strange cursor movement

konkovaa, 2020-03-26:
Hi!
First of all - thank for your great job!

I installed iichid on fresh FreeBSD 12.1 with latest XOrg package (1.20.7) on Dell 7490 and pointstick and external mouse are working as expected (cursor moving, buttons) without any configuration.
Problem is just with touchpad - there are lags beetwen start(end) move finger and cursor and cursor speed is really slow. I tryed to tune some parameters throu xorg.conf (AccelProfiles, AccelSpeed, etc.), but without success.

Do you have any idea?
Some files: [dmesg.log](https://github.com/wulf7/iichid/files/4388985/dmesg.log), [Xorg.0.log](https://github.com/wulf7/iichid/files/4388987/Xorg.0.log), [libinput.log](https://github.com/wulf7/iichid/files/4388989/libinput.log)

wulf7, 2020-03-26:
Try to upgrade your kernel to any 2020y 12-stable and rebuild iichid.

If it  does not fix your issue or you can not update kernel for some reason, try
```
$ sudo sysctl dev.iichid.0.sampling_rate_slow=60
```

konkovaa, 2020-03-27:
dev.iichid.0.sampling_rate_slow=60 didn't help in 12.1-RELEASE, so I upgraded to 12-STABLE and bingo - touchpad start to work as expected: buttons, cursor moving and (even!) two-fingers scrolling.
With upgrade to 12-STABLE I've got new input devices:
LID switch as switch and SLEEP- and POWER- buttons as keyboard ;-).
But I've lost i915drm kernel module - right after kldload it shows just black screen.
So, I've tested touchpad with xf86-video-scfb driver and now I'm thinking how to reanimate i915drm.

Anyway, thank you for your help and have a nice weekend!

closed 2020-03-27 by konkovaa