wulf7/iichid #29: Ducky keyboard, multimedia keys not recognized

gregf, 2020-04-16:
```
# dmesg | grep -i ducky
ugen2.2: <DuckyChannel International Co., Ltd. Ducky Keyboard> at usbus2
ukbd0: <DuckyChannel International Co., Ltd. Ducky Keyboard, class 0/0, rev 2.00/1.10, addr 2> on usbus2
ums1: <DuckyChannel International Co., Ltd. Ducky Keyboard, class 0/0, rev 2.00/1.10, addr 2> on usbus2
```

```
# usbconfig
ugen2.2: <DuckyChannel International Co., Ltd. Ducky Keyboard> at usbus2, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)
```

I also ran dmesg -c than ran kldload iichid there was no output.

FreeBSD fluffy.local 13.0-CURRENT FreeBSD 13.0-CURRENT #0 r359929: Tue Apr 14 16:55:34 EDT 2020     root@fluffy.local:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64

wulf7, 2020-04-16:
Did you read **usbhid** section of README? Please, rebuild iichid according to it

gregf, 2020-04-16:
I guess I screwed up. I just put the following lines in /boot/loader.conf

ig4_load="YES"
iicbus_load="YES"
iichid_load="YES"

now I have a uhid be generated. Keys don't work out of the box, but I'm able to use `usbhidaction` to get them going from this point on.

wulf7, 2020-04-16:
iichid is not a part of base system yet. You must compile and install it yourself. See README for instructions.
There is sysutils/iichid port in port collection but it will not help being rather old.
Also you can check **/boot/modules/iichid.ko** existence. **iichid_load="YES"** string does nothing if this file is absent.

gregf, 2020-04-16:
Yep, compiling from source did it. I had no idea the port was so out of date. I'm sorry if I wasted your time.

closed 2020-04-16 by gregf

wulf7, 2020-04-16:
> I had no idea the port was so out of date.

It is not so out of date. It is done on the previous stable point and next stable point is not reached yet.

Could you post output of **sudo usbhid-dump** (sysutils/usbhid-dump port provides it) command here? It is interesting to me, why the keyboard was detected as mouse by ums driver.

gregf, 2020-04-16:
```
002:002:002:DESCRIPTOR         1587046349.106921
 05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01
 95 03 75 01 81 02 95 01 75 05 81 01 C0 05 0C 09
 01 A1 01 85 02 15 00 25 01 95 12 75 01 0A 83 01
 0A 8A 01 0A 92 01 0A 94 01 09 CD 09 B7 09 B6 09
 B5 09 E2 09 EA 09 E9 0A 21 02 0A 23 02 0A 24 02
 0A 25 02 0A 26 02 0A 27 02 0A 2A 02 81 02 95 01
 75 0E 81 01 C0 05 01 09 02 A1 01 09 01 A1 00 85
 03 05 09 19 01 29 08 15 00 25 01 75 01 95 08 81
 02 05 01 09 30 09 31 16 01 80 26 FF 7F 75 10 95
 02 81 06 09 38 15 81 25 7F 75 08 95 01 81 06 05
 0C 0A 38 02 95 01 81 06 C0 C0 05 01 09 06 A1 01
 85 04 05 07 95 01 75 08 81 03 95 E8 75 01 15 00
 25 01 05 07 19 00 29 E7 81 00 C0

002:002:001:DESCRIPTOR         1587046349.113915
 06 00 FF 09 01 A1 01 09 02 15 00 26 FF 00 75 08
 95 40 81 02 09 03 15 00 26 FF 00 75 08 95 40 91
 02 C0

002:002:000:DESCRIPTOR         1587046349.119915
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 03 95 03 75 01
 05 08 19 01 29 03 91 02 95 01 75 05 91 03 95 06
 75 08 15 00 26 A4 00 05 07 19 00 29 A4 81 00 C0

000:002:001:DESCRIPTOR         1587046349.126130
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 81 03 95 06 75 08 15 00
 26 A4 00 19 00 2A A4 00 81 00 C0 05 0C 09 01 A1
 01 85 03 75 10 95 02 15 01 26 8C 02 19 01 2A 8C
 02 81 00 C0 05 01 09 80 A1 01 85 04 75 02 95 01
 15 01 25 03 09 82 09 81 09 83 81 60 75 06 81 03
 C0 06 00 FF 09 01 A1 01 85 10 75 08 95 06 15 00
 26 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09
 02 A1 01 85 11 75 08 95 13 15 00 26 FF 00 09 02
 81 00 09 02 91 00 C0

000:002:000:DESCRIPTOR         1587046349.133154
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 10
 15 00 25 01 95 10 75 01 81 02 05 01 16 01 80 26
 FF 7F 75 10 95 02 09 30 09 31 81 06 15 81 25 7F
 75 08 95 01 09 38 81 06 05 0C 0A 38 02 95 01 81
 06 C0 C0

000:003:003:DESCRIPTOR         1587046349.139118
 05 0C 09 01 A1 01 85 01 15 00 25 01 09 E9 09 EA
 75 01 95 02 81 42 09 CD 09 B5 09 B6 09 B7 09 E2
 95 05 81 06 06 FF 01 09 01 95 01 81 06 05 0C 09
 00 95 08 81 06 09 00 75 08 95 02 81 06 06 FF 07
 09 01 95 0B 81 02 09 00 95 0F 91 02 C0 05 0B 09
 05 A1 01 85 02 09 20 15 00 25 01 95 01 75 01 81
 23 95 01 75 07 81 07 C0

```

wulf7, 2020-04-16:
Thanks. It looks like typical wireless combo consisting of 3 USB devices. 1-st is AT/XTkeyboard,2-nd is mouse+multimedia keys and 3-rd is vendor device. So it should trigger attachment of ukbd, ums and uhid drivers at once.

> When I ran usbhid-dump (both times) it locked my computer up. Not sure if that is strange or not?

It detached all USB HID drivers, so you lost control. The system is not hanging. Just re-plugging of USB keyboard should be enough to get input back.