wulf7/iichid #30: sysutils/iichid-devel for faster/easier git head testing

cederom, 2020-04-16:
Hello world :-)

I can see `iichid` is under heavy development and gains user interest momentum. As there are new features added that are not yet in the release maybe it would be helpful if port `sysutils/iichid-devel` was created in order to easily allow quick and easy git head testing? :-)

Best regards :-)
Tomek

wulf7, 2020-04-16:
> I can see iichid is under heavy development

Really it has mostly achieved its main goals (HID abstraction, TLC-based bus and HID to evdev mapper) so no significant further development is planned.
I plan to do some polishing of existing code, release ver 0.0.2 and than switch to in-tree development with following deprecation of wulf7/iichid repo.
Development port is not required in this case but i can release intermediate version if someone insists.

arrowd, 2020-04-17:
iichid works flawlessly for me as for now. I'll hapilly test another intermediate release, though.

cederom, 2020-04-17:
The goal of `iichid-devel` would be to simply type `Make` in order to:
* fetch git head
* compile it easily
* install it in a proper way
* test it
* uninstall it in a clean way

Each time new commit shows up on the git repo user could simply have it working from userland with one `make` from a port directory (ports can fetch git head) no need to setup the build environment or even rebuild the kernel :-)

arrowd, 2020-04-17:
Such Makefile can be committed into this repo directly, then.

wulf7, 2020-05-05:
v 0.0.2 is released. I'll submit PR soon