wulf7/iichid #31: AmazonBasics Wired Keyboard issue

Menelkir, 2020-04-29:
Hello. The version in ports didn't worked for me, so I've sticked to the github and worked so far. Some weeks ago I had an issue with my hdd and had to reinstall freebsd. No big deal, I have backups of my known good configurations so i did my configuration again, but this time when I start sddm only the mouse works, the keyboard doesn't wok. Doesn't matter if I load the modules later or in /boot/loader.conf (I've sticked to the /etc/rc.conf for now because I can enter via single user to comment the line instead of using a usb stick to fix it). I've also created a /usr/local/etc/rc.d/usbprobe to just run usbconfig at boot to see if the keyboard re-detect with iichid without success. If I load the modules manually after the X startup, the driver loads fine without extra output.

I'll attach the files for debug.

[dmesg.txt](https://github.com/wulf7/iichid/files/4552085/dmesg.txt)
[pciconf.txt](https://github.com/wulf7/iichid/files/4552086/pciconf.txt)
[usbconfig.txt](https://github.com/wulf7/iichid/files/4552087/usbconfig.txt)
[devinfo.txt](https://github.com/wulf7/iichid/files/4552084/devinfo.txt)

Menelkir, 2020-04-29:
Update:
If I uncomment `SRCS += hskbd.c` in Makefile, it works as expected.

wulf7, 2020-04-29:
Hi! Were you running STABLE before hdd issue? Keyboard driver is not compiled on 12.1-RELEASE and requires any 2020 year stable due to KPI changes.

For now, I forbid attachment to keyboards on hkbd-less iichid, so ukbd will be used instead of it on 12.1-RELEASE. In most cases it should not harm multimedia keys support as they usually presented as different USB device. Try last revision. Please attach dmesg output one more time on success.

wulf7, 2020-04-29:
> Update: If I uncomment SRCS += hskbd.c in Makefile, it works as expected.

hskbd does not support system console input, so I don't recommend use it if you do not have AT-keyboard attached.

Menelkir, 2020-04-29:
> Hi! Were you running STABLE before hdd issue? Keyboard driver is not compiled on 12.1-RELEASE and requires any 2020 year stable due to KPI changes.

Yes, I was using 12.1-RELEASE before and it was running fine. Actually in my first try was more like experimental and since it worked, I didn't updated.
 
> For now, I forbid attachment to keyboards on hkbd-less iichid, so ukbd will be used instead of it on 12.1-RELEASE. In most cases it should not harm multimedia keys support as they usually presented as different USB device. Try last revision. Please attach dmesg output one more time on success.

I'll try it asap.

Menelkir, 2020-04-29:
> For now, I forbid attachment to keyboards on hkbd-less iichid, so ukbd will be used instead of it on 12.1-RELEASE. In most cases it should not harm multimedia keys support as they usually presented as different USB device. Try last revision. Please attach dmesg output one more time on success.

It works perfectly fine now, thanks!

[dmesg.txt](https://github.com/wulf7/iichid/files/4555127/dmesg.txt)

closed 2020-04-30 by Menelkir