wulf7/iichid #32: ACPI is not working properly

hadilq, 2020-04-29:
I'm an owner of GL553VD ROG Asus laptop and I recently migrated to FreeBSD. I installed this great driver, which helped me to work with the TouchPad. But the power manager in KDE is not recognizing the battery. Also the Fn keys are not working. So cannot change the brightness. I'm not sure if my problems are related to this repository, but let's give it a shot :) Also I believe the following information would help you as `dmesg` throws a lot of errors. Thank you for your time.
```
#devinfo -rv                                                                                                                                                                                                                           
nexus0
  cryptosoft0
  efirtc0
  apic0
      I/O memory addresses:
          0xfec00000-0xfec0001f
  ram0
      I/O memory addresses:
          0x0-0x57fff
          0x59000-0x9cfff
          0x100000-0x6d316fff
          0x6d31c000-0x79d25fff
          0x7b3fe000-0x7b3fefff
          0x100000000-0x47effffff
  aesni0
  acpi0
      Interrupt request lines:
          0x9
      I/O ports:
          0x2e-0x2f
          0x4e-0x4f
          0x61
          0x63
          0x65
          0x67
          0x70
          0x80
          0x92
          0xb2-0xb3
          0x680-0x69f
          0x800-0x87f
          0x164e-0x164f
          0x1800-0x18fe
          0xff00-0xfffe
          0xffff
      I/O memory addresses:
          0xdffe0000-0xdfffffff
          0xe0000000-0xefffffff
          0xfed10000-0xfed17fff
          0xfed18000-0xfed18fff
          0xfed19000-0xfed19fff
          0xfed20000-0xfed3ffff
          0xfed45000-0xfed8ffff
          0xfed90000-0xfed93fff
          0xfee00000-0xfeefffff
          0xff000000-0xffffffff
    acpi_ec0 pnpinfo _HID=PNP0C09 _UID=0 at handle=\_SB_.PCI0.LPCB.EC0_
        I/O ports:
            0x62
            0x66
    cpu0 pnpinfo _HID=none _UID=0 at handle=\_PR_.CPU0
      acpi_perf0
      est0
      p4tcc0
      coretemp0
      cpufreq0
    cpu1 pnpinfo _HID=none _UID=0 at handle=\_PR_.CPU1
      acpi_perf1
      est1
      p4tcc1
      coretemp1
      cpufreq1
    cpu2 pnpinfo _HID=none _UID=0 at handle=\_PR_.CPU2
      acpi_perf2
      est2
      p4tcc2
      coretemp2
[... 1517 lines cut ...]
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
iichid0: input report too big. requested=14 received=16464
```

wulf7, 2020-04-30:
> But the power manager in KDE is not recognizing the battery.

Try to use  `sudo acpiconf -i0` to query battery status. If it wont help, there are chances with tweaking BIOS settings or hw.acpi.osname loader variable.

> Also the Fn keys are not working.

There can be 2 reasons:
1. Fn keys are atkbd keys with nonstandard keycodes. That require evdev keymap patching.
2. Fn keys are ACPI events. In that case you can try to kldload whole bunch of acpi_* drivers and than catch events at devd socket with `sudo socat /var/run/devd.pipe -`. After one has been catched you can feed it to devd. See /etc/devd/asus.conf for example.

> So cannot change the brightness.

That is normal as brightness support is too quirky. You can try.
1. Kldload one of drm-kmod drivers,
2. Kldload acpi_video.ko
3. Kldload acpi_video.ko after drm-kmod driver has been kldloaded.
4. Try to catch events at devd socket after acpi_* drivers have been kldloaded. Than feed intel_backlight or sys.class.backlight. sysctls through devd.
5. Hack ACPI-notification handler in acpi_ec.c than filter out brightness events and than stream them to devd or evdev.
6. Something else.

> dmesg throws a lot of errors.

12.1-RELEASE is not recommended for use with I2C part of iichid. You should use at least 2020-year 12-STABLE

hadilq, 2020-05-17:
Thanks @wulf7! Sorry for being late! I tried some of those solutions without success, and unfortunately I don't have enough time to stick to solve them :(

closed 2020-05-17 by hadilq