wulf7/iichid #33: Full featured driver for PlayStation DualShock2 controller on IICHID?

cederom, 2020-05-05:
Hello world :-)

I have this nice PS4 controller that can work over USB and Bluetooth. It works over USB at this point on FreeBSD with using HID but not all features are supported (accelerometer, touch pad, etc, does not work).

Is it possible to extend this HID driver with additional features using this IICHID project?

How this can be achieved?

Any hints appreciated :-)

Tomek

wulf7, 2020-05-06:
> Is it possible to extend this HID driver with additional features using this IICHID project?

Certainly, yes.

> How this can be achieved?

As I do not own such a device, someone else should write support for it.

> Any hints appreciated :-)

If you are going to create this driver, the first thing you should do is to find out if report descriptor located in this device is valid or no and then write a good one if "no".
Than you should find out which events are expected by userland software to handle input.
And than you should create HID->evdev (or something else) convertor on base of iichid. xb360gp.c is a good example of such a convertor.

cederom, 2020-05-07:
Thank you @wulf7 for all of the hints!

I have last one question. Are "descriptors" those funny HEX tables that are kinf of mask what device can do? Do you have a nice reference to read about those descriptos? I downloaded once the HID specification but the implementation was still a bit unclear. From what I saw in the sources of Linux and FreeBSD HID device drivers some parts were similar and some differer probably that was the source of different capabilities availability?

wulf7, 2020-05-07:
> I have last one question. Are "descriptors" those funny HEX tables that are kinf of mask what device can do?

Yes, they are.

> Do you have a nice reference to read about those descriptos? I downloaded once the HID specification but the implementation was still a bit unclear.

The HID specs and source code of parsers are the best docs, but there are some small easy to use tools to make life better. See e.g:
1. https://eleccelerator.com/usbdescreqparser/ - it could decrypt HID report given in HEX format
2. `sudo usbhidctl -al -f /dev/hidraw*` where hidraw* can be obtained from dmesg ouput. It can dump decoded data in real-time, not only decrypt report descriptor.

You can use latter utility to check if DualShock2 controller contains good descriptor or not. Just run it and try to push all buttons one by one, touch trackpad area e.t.c. and see if dumped data matches your doings or not. If it does, you do not need to craft custom descriptor. Built-in one can be used.

> I mostly care about accelerometer that is built into the controller and I would like to use that data to drive camera in VR environment (fun project).. having a trackpad data would be nice too but optional.

Do you have any tools to process accelerometer data? If you do and this software can not process raw hid data but supports some other simple protocol like evdev, than you can send me such a device to write a driver.

cederom, 2020-05-08:
Thanks for the hints @wulf7! You rox! I will try the `usbhidctl` :-) 

I base my work on PS4 DualShock4 controller demo that was written for Blender up to 2.79. DS4/PS4 controller support was there in Blender ready to use. It does not use evdev, I guess its more USB HID. But.. it was working on Blender Game Engine that was removed in 2.80. Still Blender 2.79 can build and work on FreeBSD. Its a bit mess right now, but that could be quite nice inexpensive easy to buy controller with all available digital, analog, IMU, and even trackpad inputs :-)

https://www.sharecg.com/v/81050/related/5/3D-Model/Blender-Game-Engine-PS4-Controller-Python-Script

I am working right now on a different projects so its not high priority stuff.. but if you have some time and will to play with this kind of controller I can transfer lets say $100 via paypal so you can get it locally and see how it works with `iichid`? :-)

cederom, 2020-05-08:
plz drop me a private message to tomek at cedro dot info :-)

wulf7, 2020-05-10:
PS4 DualShock4 ... Blender . Really you have intrigued me a lot :-). Unfortunatelly, I can't just visit nearest store due to COVID-19, Could you say which one game controller should be chosen? https://www.mvideo.ru/playstation-4327/ps4-aksessuary-4334/f/category=geimpady-dlya-ps4-7667 can give you an idea what can be bought in Moscow just now. Do not try to pay for it as payment can be blocked by anti-fraud system. Just point me to the proper device. Unfortunately, all the sites about russian online trade are in russian.

cederom, 2020-05-10:
Any original PS4 controller will have the same standard features and functions. All of them just seem to have different painting :-) Just do not buy any "clone" or "replacement" as they would be totally different device inside :-) How about eBay or Amazon and international shipping? We use couriers and inpost-automats deliveries a lot now in Poland :-) Thanks!! :-)

wulf7, 2020-05-10:
I have ordered this one: https://www.mvideo.ru/products/geimpad-dlya-konsoli-ps4-playstation-4-dualshock-4-v2-black-cuh-zct2e-40065372 I am almost sure that it is not genuine Sony but compatible one.
> How about eBay or Amazon and international shipping?
Do not forget that local shipping takes one day while international takes up to the month
Moscow is twice bigger than New-York so usually you can by virtually anything you want locally.

wulf7, 2020-05-17:
Initial support is added, but there are lot things to do left,

cederom, 2020-05-18:
Thank You @wulf7 YOU ROX! =)

arrowd, 2021-01-01:
I've got a PS5 controller. Is this PR related to its support?

wulf7, 2021-01-01:
Although PS5 controller is looking pretty much like a slightly modified PS4 clone, it is better to open new issue.