wulf7/iichid #37: 32bit compat support on 64bit systems?

netchild, 2020-07-07:
Hi,

any chance to get 32bit compat support on 64bit systems?

The story behind this request is: 
 - you can run the Linux Steam client on FreeBSD (games/linux-steam-utils port)
 - [some games work](https://github.com/shkhln/linuxulator-steam-utils/wiki/Compatibility)
 - some games are 32bit, some are 64bit
 - to have gamepad support working for 32bit and 64bit games, some [pre-requisites](https://github.com/shkhln/linuxulator-steam-utils/wiki/Home-improvement-ideas) are needed, ane of them would be 32bit compat support on 64bit systems for your software.

I have already discussed with the author of webcamd to include the patch referenced on the pre-req page, and now I'm looking into linux_libusb with him. And as such this request here if there is the possibility to get you interested in 32bit compat support.

Bye,
Alexander.

wulf7, 2020-07-07:
Did you try https://svnweb.freebsd.org/base?view=revision&revision=359428 ? It was MFC-ed to 12-STABLE.

shkhln, 2020-07-16:
> https://svnweb.freebsd.org/base?view=revision&revision=359428

Somehow I missed that commit. I'm not going to test it (no iichid-compatible gamepad), but that should be enough.

wulf7, 2020-07-16:
There is no need to run iichid to test it. You can just compile evemu-record on 32bit system or with -m32 option and than run it on 64bit system specifying input/eventX device node to check if any events are coming in.
Evemu-record binary compiled under Linux could be good test tool too. But I never tested evdev that way.

wulf7, 2021-02-26:
Done long ago.

closed 2021-02-26 by wulf7