wulf7/iichid #38: Kernelmodule load order

dasTor, 2020-07-11:
First of all: thanks for the awesome work

I am currently using freebsd-stable (r363092) on a dell latitude 7280
I am loading your mod in /etc/rc.conf with:
kld_list="/boot/modules/i915kms.ko ig4 /boot/modules/iichid.ko"

when omitting the ig4 module before iichid, my touch pad won't work.
Maybe this can be fixed in iichid or ig4 or be documented.

If you need further data from me, i am happy to help.

Regards,
Daniel

wulf7, 2020-07-14:
Thanks for the report. I added line about ig4 driver to README. It is not necessary to add it to kld_list on most systems as ig4 driver is loaded automatically at the boot time through devmatch utility. But it happens successfully only with PCI-bus ig4 driver and does not work with ACPI-bus one (Intel Atoms mostly).
It is not iichid bug but FreeBSD's one, so I don't want hardcode explicit dependence on ig4 driver into iichid. Just brief line in README is enough, IMO.

dasTor, 2020-07-16:
thank you, this can be closed.

closed 2020-07-16 by dasTor