wulf7/iichid #41: Touchpad not working after resuming FreeBSD 12.1

p-zielin, 2020-08-28:
I have FreeBSD 12.1 on my Dell Latitude 5580. After closing computer's cover OS is suspended. Here is my sysctl.conf config for power management: hw.acpi.lid_switch_state=S3
After resuming system, touchpad is not working. It's freezed and sometimes mouse cursor goes out of screen.
dmesg output prints: **iichid0: read error occured: 3**
Reloading iichid module doesn't help.

p-zielin, 2020-08-28:
After iichid module is reloaded, dmesg prints:
iichid0: could not retrieve HID descriptor from the device: 3

wulf7, 2020-08-28:
I can't help here, sorry.
ig4 driver bundled with 12.1-RELEASE does not support suspend/resume. You should upgrade to 12-STABLE or 13-CURRENT.

p-zielin, 2020-08-28:
Ok, thank you.

closed 2020-08-28 by p-zielin