wulf7/iichid #42: ThinkPad X1 Carbon Gen7: trackpad randomly stops responding for some time

sreehax, 2020-10-10:
I followed the installation instructions on the README and when I kldloaded iichid my trackpad did work though not consistently. Sometimes the trackpad just stops responding to movement. Sometimes it's fixed by waiting or clicking around. Otherwise I basically have to use the trackpoint. Are there any files/info I can provide to help debug this issue? I'm running on a 12.2-STABLE compiled a few days ago.

wulf7, 2020-10-11:
Please provide:
1. Console output of kldloading iichid.ko on debug level 1 (default for wulf7/iichid, sysutils/iichid enables it with DEBUG option turned on at port compile stage).
2. Install devel/evemu, run evemu-record as root in terminal window and choose your touchpad as source. It will show touchpad events when they are coming out of kernel through /dev/input/event* character device. Do they stop while touchpad freezes?

justfly1111, 2021-10-01:
I have same issue on a Lenovo yoga from this year