wulf7/iichid #45: Not work in xiaomi 6Y30 Notebook

ykla, 2020-11-04:
How make It work?

ykla, 2020-11-04:
```
root@ykla:~ # libinput list-devices 
Device:           System keyboard multiplexer
Kernel:           /dev/input/event0
Group:            1
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           System mouse
Kernel:           /dev/input/event1
Group:            2
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

Device:           Power Button
Kernel:           /dev/input/event2
Group:            3
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           Control Method Lid Switch
Kernel:           /dev/input/event3
Group:            4
Seat:             seat0, default
Capabilities:     switch
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           AT keyboard
Kernel:           /dev/input/event4
Group:            5
Seat:             seat0, default
Capabilities:     keyboard 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   n/a
Rotation:         n/a

Device:           PixArt USB Optical Mouse, class 0/0, rev 2.00/1.00, addr 1
Kernel:           /dev/input/event5
Group:            6
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

root@ykla:~ # 

```

wulf7, 2020-11-04:
What is an OS version?
Does I2C driver (ig4.ko) is loaded ?
What is console output while kldloading iichid?

ykla, 2020-11-05:
```
root@ykla:~ # uname -a
FreeBSD ykla 12.2-RELEASE FreeBSD 12.2-RELEASE r366954 GENERIC  amd64
```
I have load ig4.ko and no output while kldloading iichid.
```
root@ykla:~ # kldload iichid
root@ykla:~ # 
```

ykla, 2020-11-05:
ok,after I kldload, it works. thanks.

closed 2020-11-05 by ykla