wulf7/iichid #46: Ignore interfaces taken by uaudio

myfreeweb, 2020-11-11:
When `uaudio` recognizes volume keys on an audio device, it does this:

```c
	/* prevent the uhid driver from attaching */
	usbd_set_parent_iface(uaa->device, sc->sc_hid.iface_index,
	    sc->sc_mixer_iface_index);
```

But that doesn't seem to prevent `usbhid` from attaching :/

With nothing done to address this issue, this results in weird behavior of these keys — as both `uaudio` and `hcons` grab events from the same endpoint, the `KEY_VOLUMEDOWN/UP` keys don't consistently get released, it's a complete mess.

I'll submit a patch making `uaudio`'s HID support optional (via sysctl), but regardless of that, `usbhid` should not attach when `uaudio` does take the interface.

wulf7, 2020-11-11:
> But that doesn't seem to prevent `usbhid` from attaching :/

Could it be a module load ordering or probe priority issue? I see nothing in `uhid` that checks `uaudio` presence.

myfreeweb, 2020-11-11:
`uaudio` manually finds the hid interface in a `usb_desc_foreach` loop, the only thing preventing `uhid` is supposed to be that `set_parent` call. The only check for the parent is in `usb_probe_and_attach_sub`, called from `usb_probe_and_attach` right after filling all the `uaa.info`. So no driver should be initialized. Weird stuff...

OH WAIT. Soooo my device has *two* HID interfaces! And `uaudio` only takes one.

myfreeweb, 2020-11-11:
Hm, this just stopped happening, now `usbhid` only attaches to the interface that wasn't taken. Weird.

closed 2020-11-11 by myfreeweb