wulf7/iichid #47: Logitech G15 G-Keys

Menelkir, 2020-11-20:
Not exactly an issue, the keyboard works pretty well (even volume and media player buttons). I just want to know if it's possible to make the G-Keys work, no macros or fancy stuff, just like the extra keys (home, mail, etc) of other keyboards.

[devinfo.txt](https://github.com/wulf7/iichid/files/5574758/devinfo.txt)
[dmesg.txt](https://github.com/wulf7/iichid/files/5574759/dmesg.txt)
[pciconf.txt](https://github.com/wulf7/iichid/files/5574760/pciconf.txt)
[usbconfig.txt](https://github.com/wulf7/iichid/files/5574761/usbconfig.txt)

wulf7, 2020-11-21:
Firstly, run `evemu-record` (devel/evemu) and see if pressing the G-keys results in any events
If not, then try `sudo usbhidctl -f /dev/hidraw# -l -a`. Where # is a you keyboard`s device node.

Menelkir, 2020-11-21:
Nope. None of them respond to the commands.

wulf7, 2020-11-22:
> Nope. Tried both commands and the keys doesn't work.

Most probably, G-keys belong to other USB device. As your usbconfig output does not give clear answer to that question, try to replug keyboard and if more than one device appeared reattached in console output, run evemu-record and usbhidctl with this **other** device.

Menelkir, 2020-11-23:
I've tried with all the devices (that G15 makes available) without luck, any other suggestions?

wulf7, 2020-11-23:
> any other suggestions

According to Linux driver, G-keys require extra initialization: https://elixir.bootlin.com/linux/v5.10-rc5/source/drivers/hid/hid-lg-g15.c#L781 and a special support.

Could you provide some additional info about your device. Just for the record:

1. HID report descriptor: run `usbhid-dump` from `sysutils/usbhid-dump` port
2. USB descriptors: run `usbconfig dump_all_desc`

Menelkir, 2020-11-24:
[usbconfig.txt](https://github.com/wulf7/iichid/files/5586835/usbconfig.txt)
[usbhid-dump.txt](https://github.com/wulf7/iichid/files/5586836/usbhid-dump.txt)

Menelkir, 2021-01-25:
I know is not directly related, but I've managed to make g15daemon work on freebsd, the screen works (even g15stats) but the keys are still a missing piece. The linux driver for this is somehow newer than g15daemon, so maybe is something I'm missing in g15daemon?
My port is [here](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=252987) if you're interested.

wulf7, 2021-01-25:
> but the keys are still a missing piece.

I think uinput plugin is required for keys to work.