wulf7/iichid #51: touchpad on Panasonic CF-QV doesn't work

omatsuda, 2021-01-03:
Thank you very much for developing iichid.

I recently set up FreeBSD for a Panasonic laptop CF-QV.  The touch pad of this machine is not recognized as a PS/2 mouse and definitely we need iichid.  But at this moment, it is not successful: once the mouse cursor is moved with a short stroke on the touch pad, the cursor keeps moving even after the finger is off from the touch pad.

I'm using 13.0-CURRENT r368820 (right after the snapshot 201231) and the latest iichid (Dec. 27).

"dmesg" (dmesg.txt) shows iicbus[01] and iic[01] are created, though there are some anxious messages:

iicbus0: <unknown card> at addr 0x2c
iicbus1: <unknown card> at addr 0x10
iicbus1: <unknown card> at addr 0x1f
iicbus1: <unknown card> at addr 0x11

Anyway, when "kldload iichid", hms0 is created, but not hmt0. hms0 is connected to /dev/input/event6 (kldload.txt)

Output of "evemu-record /dev/input/event6" is attached. (event6.txt) Here I first press the left button and then put a short stroke on the pad. hms0 keeps reporting the (relative) cursor position even after the stroke motion ends. The report stops when another press of the left button is given at the end of the event log.  It seems spurious reports are continuously generated even the finger is off the touch pad.

Is there any way to correct this?

[dmesg.txt](https://github.com/wulf7/iichid/files/5761940/dmesg.txt)
[kldload.txt](https://github.com/wulf7/iichid/files/5761945/kldload.txt)
[event6.txt](https://github.com/wulf7/iichid/files/5761946/event6.txt)

wulf7, 2021-01-04:
> Here I first press the left button and then put a short stroke on the pad. hms0 keeps reporting the (relative) cursor position even after the stroke motion ends. The report stops when another press [...]

It is a result of missing GPIO interrupt support. Sampling code reads last report generated by touchpad over and over again that results in cursor drift.

> Is there any way to correct this?

No way with current code. I'll try to write code to filter out consecutive repeated packets when iichid is running in sampling mode. Hope it will help, so stay tuned.

> In addition, if it is recongnized as a hmt device, it would be super.

Unfortunately, the touchpad uses proprietary Synaptics protocol rather than Microsoft one, thus it is not compatible with hms.
I do have one device that uses such a protocol but it is connected over SMBus rather over I2C so even if would port driver for it, that driver could not be directly used.

cederom, 2021-01-04:
Hey there :-) I have Panasonic CF-MX4 and CF-C2. The first one offers only basic Touchpad functionality like mouse movement, but no Trackpad functionalisies like two finger scroll, tap, zoom, etc. It works well on Linux and Windoze. It is not the I2C device but I would love to make it work in a Synaptics Trackpad way :-)

omatsuda, 2021-01-06:
Thank you for your reply, @wulf7 -san.

I understood the situation. One possibly related observation is that the drift speed and direction correspond to those of the latest stroke on the touchpad, which agree with your description.

I am looking forward to hearing from you about new filtering code.

wulf7, 2021-01-08:
> I am looking forward to hearing from you about new filtering code.

The code is added. Try it, @omatsuda -san

wulf7, 2021-01-08:
> Hey there :-) I have Panasonic CF-MX4 and CF-C2. The first one is the one I really love and work on it everyday, the second one is my working machine, both offers only basic Touchpad functionality like mouse movement, but no Trackpad functionalisies like two finger scroll, tap, zoom, etc. These Synaptics touchpads work well on Linux and Windoze. It is not the I2C device but I would love to make it work in that fancy Apple Trackpad way :-)

Hi @cederom -san, :-)
Do you know the name of the bus, your touchpad is attached with?
You can run e.g. `# libinput debug-events` and look for name of device that emits mouse events. It allows us to get a driver name.

cederom, 2021-01-09:
@wulf7 -san =) The output is below =)
```
% libinput debug-events
(..)
-event5   DEVICE_ADDED     Generic PS/2 mouse                seat0 default group6  cap:p left scroll-nat scroll-button
(..)
 event5   POINTER_MOTION   +0.968s        2.59/ -7.77 ( +2.00/ -6.00)
 event5   POINTER_MOTION   +0.980s        2.57/ -7.71 ( +2.00/ -6.00)
 event5   POINTER_MOTION   +0.988s        3.75/ -6.25 ( +3.00/ -5.00)
 event5   POINTER_MOTION   +0.999s        2.46/ -6.15 ( +2.00/ -5.00)
 event5   POINTER_MOTION   +1.008s        3.40/ -4.54 ( +3.00/ -4.00)
 event5   POINTER_MOTION   +1.019s        3.24/ -3.24 ( +3.00/ -3.00)
 event5   POINTER_MOTION   +1.028s        1.00/ -2.00 ( +1.00/ -2.00)
 event5   POINTER_MOTION   +1.039s        2.00/  0.00 ( +2.00/ +0.00)
 event5   POINTER_MOTION   +1.047s        1.00/  0.00 ( +1.00/ +0.00)
 event5   POINTER_MOTION   +1.079s        0.00/  0.94 ( +0.00/ +1.00

% libinput list-devices
Device:           Generic PS/2 mouse
Kernel:           /dev/input/event5
Group:            6
Seat:             seat0, default
Capabilities:     pointer
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

% uname -a
FreeBSD 0xCFMX4 12.2-RELEASE-p1 FreeBSD 12.2-RELEASE-p1 GENERIC  amd64
```

omatsuda, 2021-01-09:
Thank you very much!
Your new code works perfectly!
Now I don't see any drift.

> The code is added. Try it, @omatsuda -san

wulf7, 2021-01-09:
> % libinput debug-events
> (..)
> -event5   DEVICE_ADDED     Generic PS/2 mouse                seat0 default group6  cap:p left scroll-nat scroll-button
> (..)

So, it is PS/2. Could you do the same after booting from any recent Linux live distro?
If the touchpad is detected properly, it's name will begin with XxxxxPS/2. What is Xxxxx? Definitely, you are out luck if Xxxxx is "Alps". FreeBSD does not support PS/2 Alps and most probably will never do as there are too much versions of it exist. (Linux driver supports 9(!) variants of Alps PS/2 protocol). But if Xxxxx is not "Alps" there are chances to fix it relatively easy.

cederom, 2021-01-09:
Thank you @wulf7 this is a Synaptics PS/2 device :-) But the hw.psm.synaptics does not work with it.. I gave up.. is there any chance to make it work?? :-)

wulf7, 2021-01-10:
If you sure, that it is Synaptics, set debug.psm.loglevel=5 in /boot/loader.conf and post boot log here.

cederom, 2021-01-14:
It is something like this:
```
psm: ENABLE_DEV return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 20 03 64
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_RESOLUTION (2) 00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 20 03 64
psmintr: 38 f1 f5 00 00 00
psmintr: Sync bytes now 0000,0000
psmintr: 38 f1 f8 00 00 00
psmintr: 38 f2 fb 00 00 00
psmintr: 38 fd fd 00 00 00
psmintr: 38 f6 fe 00 00 00
psmintr: 18 f5 00 00 00 00
psmintr: 18 f4 00 00 00 00
psmintr: 18 f4 00 00 00 00
psmintr: 18 f4 04 00 00 00
psmintr: 18 f5 05 00 00 00
psmintr: 18 f6 06 00 00 00
psmintr: 18 f7 08 00 00 00
psmintr: 18 f8 08 00 00 00
psmintr: 18 f9 0a 00 00 00
psmintr: 18 fa 0a 00 00 00
psmintr: 18 fb 0c 00 00 00
psmintr: 18 fd 0a 00 00 00
psmintr: 18 fd 0a 00 00 00
psmintr: 18 fe 0b 00 00 00
psmintr: 08 00 09 00 00 00
psmintr: 08 00 06 00 00 00
psmintr: 08 00 06 00 00 00
psmintr: 08 00 06 00 00 00
psmintr: 08 00 05 00 00 00
psmintr: 08 00 03 00 00 00
psmintr: 08 00 04 00 00 00
```

wulf7, 2021-01-14:
That is not I expected to see. Just reboot and get a `dmesg` output. It should be rather big as psm driver will try to probe more than 10 different PS/2 device types.

cederom, 2021-01-14:
Here you go sir :-)
```
Jan 14 22:35:02 0xCFMX4 kernel: ---<<BOOT>>---
Jan 14 22:35:02 0xCFMX4 kernel: Copyright (c) 1992-2020 The FreeBSD Project.
Jan 14 22:35:02 0xCFMX4 kernel: Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
Jan 14 22:35:02 0xCFMX4 kernel: 	The Regents of the University of California. All rights reserved.
Jan 14 22:35:02 0xCFMX4 kernel: FreeBSD is a registered trademark of The FreeBSD Foundation.
Jan 14 22:35:02 0xCFMX4 kernel: FreeBSD 12.2-RELEASE-p1 GENERIC amd64
Jan 14 22:35:02 0xCFMX4 kernel: FreeBSD clang version 10.0.1 (git@github.com:llvm/llvm-project.git llvmorg-10.0.1-0-gef32c611aa2)
Jan 14 22:35:02 0xCFMX4 kernel: VT(vga): resolution 640x480
Jan 14 22:35:02 0xCFMX4 kernel: CPU: Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz (2294.74-MHz K8-class CPU)
Jan 14 22:35:02 0xCFMX4 kernel:   Origin="GenuineIntel"  Id=0x306d4  Family=0x6  Model=0x3d  Stepping=4
Jan 14 22:35:02 0xCFMX4 kernel:   Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
Jan 14 22:35:02 0xCFMX4 kernel:   Features2=0x7ffafbff<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,SMX,EST,TM2,SSSE3,SDBG,FMA,CX16,xTPR,PDCM,PCID,SSE4.1,SSE4.2,x2APIC,MOVBE,POPCNT,TSCDLT,AESNI,XSAVE,OSXSAVE,AVX,F16C,RDRAND>
Jan 14 22:35:02 0xCFMX4 kernel:   AMD Features=0x2c100800<SYSCALL,NX,Page1GB,RDTSCP,LM>
Jan 14 22:35:02 0xCFMX4 kernel:   AMD Features2=0x121<LAHF,ABM,Prefetch>
Jan 14 22:35:02 0xCFMX4 kernel:   Structured Extended Features=0x21c2fbb<FSGSBASE,TSCADJ,BMI1,HLE,AVX2,SMEP,BMI2,ERMS,INVPCID,RTM,NFPUSG,RDSEED,ADX,SMAP,PROCTRACE>
Jan 14 22:35:02 0xCFMX4 kernel:   Structured Extended Features3=0x9c000400<MD_CLEAR,IBPB,STIBP,L1DFL,SSBD>
Jan 14 22:35:02 0xCFMX4 kernel:   XSAVE Features=0x1<XSAVEOPT>
Jan 14 22:35:02 0xCFMX4 kernel:   VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID
Jan 14 22:35:02 0xCFMX4 kernel:   TSC: P-state invariant, performance statistics
Jan 14 22:35:02 0xCFMX4 kernel: real memory  = 8606711808 (8208 MB)
Jan 14 22:35:02 0xCFMX4 kernel: avail memory = 8173576192 (7794 MB)
Jan 14 22:35:02 0xCFMX4 kernel: ACPI APIC Table: <MATBIO CFMX4-1 >
Jan 14 22:35:02 0xCFMX4 kernel: FreeBSD/SMP: Multiprocessor System Detected: 4 CPUs
Jan 14 22:35:02 0xCFMX4 kernel: FreeBSD/SMP: 1 package(s) x 2 core(s) x 2 hardware threads
Jan 14 22:35:02 0xCFMX4 kernel: random: unblocking device.
Jan 14 22:35:02 0xCFMX4 kernel: ioapic0 <Version 2.0> irqs 0-39 on motherboard
Jan 14 22:35:02 0xCFMX4 kernel: Launching APs: 1 3 2
Jan 14 22:35:02 0xCFMX4 kernel: Timecounter "TSC-low" frequency 1147368436 Hz quality 1000
Jan 14 22:35:02 0xCFMX4 kernel: Cuse v0.1.36 @ /dev/cuse
Jan 14 22:35:02 0xCFMX4 kernel: random: entropy device external interface
Jan 14 22:35:02 0xCFMX4 kernel: 000.000018 [4336] netmap_init               netmap: loaded module
Jan 14 22:35:02 0xCFMX4 kernel: [ath_hal] loaded
Jan 14 22:35:02 0xCFMX4 kernel: module_register_init: MOD_LOAD (vesa, 0xffffffff81115e40, 0) error 19
Jan 14 22:35:02 0xCFMX4 kernel: random: registering fast source Intel Secure Key RNG
Jan 14 22:35:02 0xCFMX4 kernel: random: fast provider: "Intel Secure Key RNG"
Jan 14 22:35:02 0xCFMX4 kernel: kbd1 at kbdmux0
Jan 14 22:35:02 0xCFMX4 kernel: nexus0
Jan 14 22:35:02 0xCFMX4 kernel: vtvga0: <VT VGA driver> on motherboard
Jan 14 22:35:02 0xCFMX4 kernel: cryptosoft0: <software crypto> on motherboard
Jan 14 22:35:02 0xCFMX4 kernel: aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS> on motherboard
Jan 14 22:35:02 0xCFMX4 kernel: acpi0: <MATBIO CFMX4-1> on motherboard
Jan 14 22:35:02 0xCFMX4 kernel: acpi0: Power Button (fixed)
Jan 14 22:35:02 0xCFMX4 kernel: unknown: I/O range not supported
Jan 14 22:35:02 0xCFMX4 kernel: cpu0: <ACPI CPU> on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: Timecounter "HPET" frequency 14318180 Hz quality 950
Jan 14 22:35:02 0xCFMX4 kernel: Event timer "HPET" frequency 14318180 Hz quality 550
Jan 14 22:35:02 0xCFMX4 kernel: Event timer "HPET1" frequency 14318180 Hz quality 440
Jan 14 22:35:02 0xCFMX4 kernel: Event timer "HPET2" frequency 14318180 Hz quality 440
Jan 14 22:35:02 0xCFMX4 kernel: Event timer "HPET3" frequency 14318180 Hz quality 440
Jan 14 22:35:02 0xCFMX4 kernel: Event timer "HPET4" frequency 14318180 Hz quality 440
Jan 14 22:35:02 0xCFMX4 kernel: atrtc0: <AT realtime clock> port 0x70-0x77 irq 8 on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: atrtc0: registered as a time-of-day clock, resolution 1.000000s
Jan 14 22:35:02 0xCFMX4 kernel: attimer0: <AT timer> port 0x40-0x43,0x50-0x53 irq 0 on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: Timecounter "i8254" frequency 1193182 Hz quality 0
Jan 14 22:35:02 0xCFMX4 kernel: Event timer "i8254" frequency 1193182 Hz quality 100
Jan 14 22:35:02 0xCFMX4 kernel: Timecounter "ACPI-fast" frequency 3579545 Hz quality 900
Jan 14 22:35:02 0xCFMX4 kernel: acpi_timer0: <24-bit timer at 3.579545MHz> port 0x1808-0x180b on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: acpi_ec0: <Embedded Controller: GPE 0xa> port 0x62,0x66 on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0
Jan 14 22:35:02 0xCFMX4 kernel: pci0: <ACPI PCI bus> on pcib0
Jan 14 22:35:02 0xCFMX4 kernel: vgapci0: <VGA-compatible display> port 0xf000-0xf03f mem 0xf6000000-0xf6ffffff,0xe0000000-0xefffffff irq 16 at device 2.0 on pci0
Jan 14 22:35:02 0xCFMX4 kernel: vgapci0: Boot video device
[... cut ...]
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 28 00 fc 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 28 00 fb 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 fe fb 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 fd f9 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 fb fa 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 fb fb 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 fb fa 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 f9 fc 00 00 00
Jan 14 22:35:06 0xCFMX4 kernel: psmintr: 38 f8 fc 00 00 00
..
```

wulf7, 2021-01-14:
It looks like broken local configuration

Do you have `hint.psm.0.flags` set in `/boot/device.hints` ?

cederom, 2021-01-14:
Hmm, maybe leftover from my testing?
```
hint.psm.0.at="atkbdc"
hint.psm.0.irq="12"
debug.psm.loglevel="1"
hint.psm.0.debug="1"
hint.psm.0.flags="0x7724"
```

wulf7, 2021-01-14:
> ```
> hint.psm.0.flags="0x7724"
> ```
You absolutely need to remove this line

cederom, 2021-01-14:
After removing this line we have:
```
Jan 14 23:54:35 0xCFMX4 kernel: atkbdc0: <Keyboard controller (i8042)> port 0x60,0x64 irq 1 on acpi0
Jan 14 23:54:35 0xCFMX4 kernel: atkbd0: <AT Keyboard> irq 1 on atkbdc0
Jan 14 23:54:35 0xCFMX4 kernel: kbd0 at atkbd0
Jan 14 23:54:35 0xCFMX4 kernel: atkbd0: [GIANT-LOCKED]
Jan 14 23:54:35 0xCFMX4 kernel: psm0: current command byte:0065
Jan 14 23:54:35 0xCFMX4 kernel: psm: ENABLE_DEV return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: DISABLE_DEV return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 00 02 64
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: device ID: 0000
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (0) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SCALING11 return code:00fa
Jan 14 23:54:35 0xCFMX4 syslogd: last message repeated 2 times
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 00 02 64
Jan 14 23:54:35 0xCFMX4 kernel: aux_mux: ping port 0
Jan 14 23:54:35 0xCFMX4 kernel: psm: ENABLE_DEV return code:00fe
Jan 14 23:54:35 0xCFMX4 kernel: aux_mux: ping port 1
Jan 14 23:54:35 0xCFMX4 kernel: psm: ENABLE_DEV return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: DISABLE_DEV return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: aux_mux: ping port 2
Jan 14 23:54:35 0xCFMX4 kernel: psm: ENABLE_DEV return code:00fe
Jan 14 23:54:35 0xCFMX4 kernel: aux_mux: ping port 3
Jan 14 23:54:35 0xCFMX4 kernel: psm: ENABLE_DEV return code:00fe
Jan 14 23:54:35 0xCFMX4 kernel: Active Multiplexing PS/2 controller v1.0 with 1 active port(s)
Jan 14 23:54:35 0xCFMX4 kernel: psm: ENABLE_DEV return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: DISABLE_DEV return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (3) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SCALING11 return code:00fa
Jan 14 23:54:35 0xCFMX4 syslogd: last message repeated 2 times
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 00 00 64
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (3) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SCALING11 return code:00fa
Jan 14 23:54:35 0xCFMX4 syslogd: last message repeated 2 times
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 00 00 64
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SCALING11 return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (0) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (3) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (2) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (1) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (3) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (1) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (2) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (3) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_DATA return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: data 00 00 00
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (200) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (100) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (80) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: device ID: 0000
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (200) 00fa
Jan 14 23:54:35 0xCFMX4 syslogd: last message repeated 1 times
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (80) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: device ID: 0000
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (200) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (100) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (80) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (60) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (40) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (20) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: device ID: 0000
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
[... 20 lines cut ...]
Jan 14 23:54:35 0xCFMX4 syslogd: last message repeated 2 times
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 10 00 64
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: device ID: 0000
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (10) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_DEV_ID return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: device ID: 0000
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (2) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (100) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SCALING11 return code:00fa
Jan 14 23:54:35 0xCFMX4 syslogd: last message repeated 3 times
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 10 00 64
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_RESOLUTION (2) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SAMPLING_RATE (100) 00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SET_SCALING11 return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: SEND_AUX_DEV_STATUS return code:00fa
Jan 14 23:54:35 0xCFMX4 kernel: psm: status 10 00 64
Jan 14 23:54:35 0xCFMX4 kernel: psm0: <PS/2 Mouse> irq 12 on atkbdc0
Jan 14 23:54:35 0xCFMX4 kernel: psm0: [GIANT-LOCKED]
Jan 14 23:54:35 0xCFMX4 kernel: psm0: model Generic PS/2 mouse, device ID 0-00, 2 buttons
Jan 14 23:54:35 0xCFMX4 kernel: psm0: config:00000000, flags:00000008, packet size:3
Jan 14 23:54:35 0xCFMX4 kernel: psm0: syncmask:c0, syncbits:00
```

wulf7, 2021-01-14:
try insert `hw.psm.mux_disabled=1` in to `/boot/loader.conf` and reboot than

wulf7, 2021-01-15:
It is appeared that SYN0503 is SMBus touchpad with PS/2 - compatibility mode.

cederom, 2021-01-15:
We have now this for `dmesg | grep psm`:
```
psm0: current command byte:0065
psm: ENABLE_DEV return code:00fa
psm: DISABLE_DEV return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 00 00 64
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_RESOLUTION (0) 00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 00 00 64
psm: SET_RESOLUTION (3) 00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 00 00 64
psm: SET_RESOLUTION (3) 00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 00 00 64
psm: SET_SCALING11 return code:00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (3) 00fa
psm: SET_RESOLUTION (2) 00fa
psm: SET_RESOLUTION (1) 00fa
psm: SET_RESOLUTION (3) 00fa
psm: SET_RESOLUTION (1) 00fa
psm: SET_RESOLUTION (2) 00fa
psm: SET_RESOLUTION (3) 00fa
psm: SEND_AUX_DEV_DATA return code:00fa
psm: data 00 00 00
psm: SET_SAMPLING_RATE (200) 00fa
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_SAMPLING_RATE (80) 00fa
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_SAMPLING_RATE (200) 00fa
psm: SET_SAMPLING_RATE (200) 00fa
psm: SET_SAMPLING_RATE (80) 00fa
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_SAMPLING_RATE (200) 00fa
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_SAMPLING_RATE (80) 00fa
psm: SET_SAMPLING_RATE (60) 00fa
psm: SET_SAMPLING_RATE (40) 00fa
psm: SET_SAMPLING_RATE (20) 00fa
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_SCALING11 return code:00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 00 00 64
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 00 00 64
psm: SET_SAMPLING_RATE (200) 00fa
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_SAMPLING_RATE (80) 00fa
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_SCALING21 return code:00fa
psm: SET_SCALING21 return code:00fa
psm: SET_SCALING21 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 10 00 64
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_SAMPLING_RATE (10) 00fa
psm: SEND_DEV_ID return code:00fa
psm: device ID: 0000
psm: SET_RESOLUTION (2) 00fa
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 10 00 64
psm: SET_RESOLUTION (0) 00fa
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 10 00 64
psm0: <PS/2 Mouse> irq 12 on atkbdc0
psm0: [GIANT-LOCKED]
psm0: model Generic PS/2 mouse, device ID 0-00, 2 buttons
psm0: config:00000000, flags:00000008, packet size:3
psm0: syncmask:c0, syncbits:00
psm: ENABLE_DEV return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 30 00 64
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 30 00 64
psmintr: 08 0d 05 00 00 00
psmintr: Sync bytes now 00c0,0000
psmintr: 08 0b 02 00 00 00
psmintr: 08 08 00 00 00 00
psmintr: 28 04 fc 00 00 00
psmintr: 28 01 fa 00 00 00
psmintr: 28 00 f6 00 00 00
psmintr: 38 fa f5 00 00 00
..
```

And this for  `sysctl -a | grep psm`:
```
device	psm
kern.evdev.input.5.phys: psm0
debug.psm.pkterrthresh: 2
debug.psm.usecs: 500000
debug.psm.secs: 0
debug.psm.errusecs: 0
debug.psm.errsecs: 2
debug.psm.hz: 20
debug.psm.loglevel: 5
irq12: psm0:23 @cpu0(domain0): 4131
hw.psm.mux_disabled: 1
hw.psm.elantech_support: 1
hw.psm.trackpoint_support: 1
hw.psm.synaptics_support: 1
hw.psm.tap_timeout: 125000
hw.psm.tap_threshold: 25
hw.psm.tap_enabled: -1
dev.psm.0.%parent: atkbdc0
dev.psm.0.%pnpinfo: 
dev.psm.0.%location: 
dev.psm.0.%driver: psm
dev.psm.0.%desc: PS/2 Mouse
dev.psm.%parent: 
dev.psmcpnp.0.%parent: acpi0
dev.psmcpnp.0.%pnpinfo: _HID=SYN0503 _UID=0
dev.psmcpnp.0.%location: handle=\_SB_.PCI0.LPCB.PS2M
dev.psmcpnp.0.%driver: psmcpnp
dev.psmcpnp.0.%desc: PS/2 mouse port
dev.psmcpnp.%parent: 
```

Still no gestures support :-(

wulf7, 2021-01-16:
> Still no gestures support :-(

It is quite possible that this touchpad can enable multitouch only over SMBus.

One thing you can try is to probe touchpad with active multiplexing mode enabled, like Linux does.
Replace 2 with 1 in following snippet. It is around line 6282 of sys/dev/atkbdc/psm.c and rebuild kernel.
Don't forget to remove hw.psm.mux_disabled=1 from /boot/loader.conf
```
        /* psm has a special support for GenMouse + SynTouchpad combination */
        if (active_ports_count >= 2) {
                for (port = 0; port < KBDC_AUX_MUX_NUM_PORTS; port++) {
```

cederom, 2021-01-30:
Hello @wulf7 and thank you for hints! It took me a white to rebuild the system as 12.2-RELEASE does not build the release at the moment, so I created a 12-STABLE memstick to boot from with suggested changes.

Because this CF-MX4 seems different and more compilicated task I have created a dedicated issue https://github.com/wulf7/iichid/issues/53 so we can leave this CF-QV thread that itself seems fixed and closed :-)

omatsuda, 2021-02-14:
I'm glad to see iichid is now included in 13-RELEASE/STABLE.  
It's a time to close this thread.  Thank you very much for all your work.

closed 2021-02-14 by omatsuda