wulf7/iichid #52: ELAN0501 Trackpad choppy after ACPI S3 sleep

probonopd, 2021-01-14:
ELAN0501 Trackpad is working well until the computer goes to sleep in ACPI S3.
After waking up, the trackpad is functional but choppy.
Some logs at https://github.com/helloSystem/ISO/issues/100, please let me know whether I should provide more.

wulf7, 2021-01-14:
I2C bus driver on 12.1-RELEASE does not support suspend/resume, interrupts and so on. You should upgrade to 12.2

probonopd, 2021-01-15:
Thank you. I will retest on 13-CURRENT.

closed 2021-01-15 by probonopd

cederom, 2021-01-15:
@probonopd you can still work with that broken driver on 12.2 with Xfce4 or when you switch off OpenGL render (as I did in Enligthenment) :-) 13-CURRENT tends to be really unstable for work when you update frequently things may crash and block your workstation this is why I have returned to RELEASE :-) You may also want to try 12-STABLE first? :-)

wulf7, 2021-01-16:
> 12.2 currently has a broken Intel GPU driver.

It is not broken. Packaged version is built for 12.1 which has LKPI subsystem binary incompatible with 12.2.
All you need is rebuild drm-kmod using ports.

probonopd, 2021-01-17:
That is what I mean by "broken" - you can't install 12.2 without recompiling stuff and expect Intel GPUs to work. Reson enough for me to stay away from that release until it is fixed.

cederom, 2021-01-17:
I know what you mean and I feel similar :-) All 12 is like this ;-) Not sure if 13 will be better. All GPU stuff is turned upside down and comes from Linux DRM implementation. Its like 20 years regression. Not to mention Wayland vs X where most stuff is hardcoded for Linux SystemD. Most people think Open-Source is Linux ;-)

If you want 12 then on every minor upgrade you need to recompile two ports to make things work (here too its not only drm-kmod but also drm-kmod-fbsd12.1 note 12.1 not 12 or 12.2 hahah).

Kind of mess :-) "Change is good" or "The only constant is change" and we have results ;-) Sand castles without a solid fundament and no one seems to care :-)

wulf7, 2021-01-17:
> All GPU stuff is turned upside down and comes from Linux DRM implementation. Its like 20 years regression.

There was no BSD DRM implementation. Even nvidia native driver is actually a Linux driver with BSD-shim wrapper.
The main problem with current DRM that BSD-shim is still GPL-led so it can not be included in to base just now and DRM drivers have to be compiled out of base. Unfortunately, it looks like deGPLization will not be finished till 13-STABLE is branched, that means DRM will remain out-if-base thing.