wulf7/iichid #53: touchpad on Panasonic Toughbook CF-MX4

cederom, 2021-01-30:
Hello world :-)

As we started working on CF-MX4 touchpad in a similar related issue https://github.com/wulf7/iichid/issues/51 but this does not look like a trivial task I have created a dedicated Issue :-)

Getting back to @wulf7 suggestion:

> It is quite possible that this touchpad can enable multitouch only over SMBus.
> 
> One thing you can try is to probe touchpad with active multiplexing mode enabled, like Linux does.
> Replace 2 with 1 in following snippet. It is around line 6282 of sys/dev/atkbdc/psm.c and rebuild kernel.
> Don't forget to remove hw.psm.mux_disabled=1 from /boot/loader.conf
> 
> ```
>         /* psm has a special support for GenMouse + SynTouchpad combination */
>         if (active_ports_count >= 2) {
>                 for (port = 0; port < KBDC_AUX_MUX_NUM_PORTS; port++) {
> ```

So I have c
[psm-panasonic-cfmx4-dmesg.txt](https://github.com/wulf7/iichid/files/5898021/psm-panasonic-cfmx4-dmesg.txt)
[psm-panasonic-cfmx4-dmesg-full.txt](https://github.com/wulf7/iichid/files/5898022/psm-panasonic-cfmx4-dmesg-full.txt)
[psm-panasonic-cfmx4-sysctl.txt](https://github.com/wulf7/iichid/files/5898023/psm-panasonic-cfmx4-sysctl.txt)

hanged that psm.c driver recompiled a system (12-STABLE as 12.2-RELEAE currently does not build), created a memstick image to boot from. I did that but there seems to be no difference. Log files are attached.

cederom, 2021-01-30:
I have two CF-MX4 laptops (they are really cool and cheap second hand), so I can now test both of them :-)

12.2-RELEASE (my main workstation):
```
psm0: <PS/2 Mouse> irq 12 on atkbdc0
psm0: [GIANT-LOCKED]
psm0: model Generic PS/2 mouse, device ID 0-00, 2 buttons
psm0: config:00000000, flags:00000008, packet size:3
psm0: syncmask:c0, syncbits:00
psm: status 30 02 64
psm: status 30 02 64
```

Modified 12-STABLE (with `if (active_ports_count >= 1)` patch):
```
psm0: <PS/2 Mouse> irq 12 on atkbdc0
psm0: [GIANT-LOCKED]
psm0: model Synaptics Touchpad, device ID 0-00, 3 buttons
psm0: config:00004000, flags:00000008, packet size:3
psm0: syncmask:00, syncbits:00
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_RESOLUTION (1) 00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 24 4b c5
psm: ENABLE_DEV return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 80 20 02
psm: SET_SAMPLING_RATE (100) 00fa
psm: SET_RESOLUTION (0) 00fa
psm: SET_SCALING11 return code:00fa
psm: SEND_AUX_DEV_STATUS return code:00fa
psm: status 20 02 64
```

Wow it seems that this patch made device recognized as Synaptics! I will have to recompile 12.2-RELEASE kernel with this patch and let you know :-)

cederom, 2021-01-30:
WHOAH! ITS ALIVE! ALIVE!!! =)

```
dev.psm.0.%parent: atkbdc0
dev.psm.0.%pnpinfo:
dev.psm.0.%location:
dev.psm.0.%driver: psm
dev.psm.0.%desc: PS/2 Mouse
dev.psm.%parent:
dev.psmcpnp.0.%parent: acpi0
dev.psmcpnp.0.%pnpinfo: _HID=SYN0503 _UID=0
dev.psmcpnp.0.%location: handle=\_SB_.PCI0.LPCB.PS2M
dev.psmcpnp.0.%driver: psmcpnp
dev.psmcpnp.0.%desc: PS/2 mouse port
dev.psmcpnp.%parent:
```

Needs some configuration but it works! You are the greatest of the master wizards of the known universe @wulf7 ! Thank You!! =)

cederom, 2021-01-30:
It even works (and works better) on Xorg without `xf86-input-synaptics` driver :-)

The natural scroll `sysctl` does not seem to work so I found that trick on forums:
```
echo 'pointer = 1 2 3 4 5 6 7 8 9 10' > ~/.xmodmap
xmodmap ~/.xmodmap
```

We need to put that patch into upstream! I got back my joy of working with a laptop! :-)

cederom, 2021-02-06:
The correct way to handle Touchpad configuration is LIBINPUT, not really moused, xinput, or xmodmap. Use `/usr/local/share/X11/xorg.conf.d/40-libinput.conf` like:
```
Section "InputClass"
        Identifier "libinput touchpad catchall"
        MatchIsTouchpad "on"
        MatchDevicePath "/dev/input/event5"
        Option "NaturalScrolling" "on"
        Option "Tapping" "on"
        Option "ClickMethod" "clickfinger"
        Driver "libinput"
EndSection
```
You can update the main readme :-)

cederom, 2021-07-07:
Hello @wulf7 any chance to fix the psm driver in FreeBSD? I have upgraded to 13.0 and still trackpad gestures does not work on my Panasonic Toughbook CF-MX4 :-(

wulf7, 2021-07-07:
Hi @cederom,

Could you follow instructions from https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=253279 ?

cederom, 2022-02-02:
The issue was fixed and landed into 13-STABLE. It should get into a RELEASE soon I hope :-)

TANK YOU @wulf7 !!! :-)

closed 2022-02-02 by cederom