wulf7/iichid #59: Help with Cooler Master MasterKeys M Multimedia Keys

goyox86, 2021-04-26:
Hi @wulf7 ! 

First of all thank you for all the effort on building these drivers and all the work I see you put supporting users.

I have been struggling with making my multimedia keys work.

I don't know much about the USB stack, HID and how all the drivers and userspace libraries fit together, so I am a bit lost.

`ukbd` seems to work perfect for anything else than the multimedia/double function keys (The ones that require `Fn` to be pressed).

I've tried multiple things initially I tried loading `hkbd`, `hcons`, following the instructions in their respective man pages (I am using the drivers shipped with Freebsd 14-CURRENT now). 

At the beginning I was thinking that `ukbd` was somehow "stealing the keyboard" I have tried to remove `ukbd` from the kernel and also preventing `devmatch` to auto load the modules. As expected no keyboard input events are received at when loading the `h*` and ukbd not being present.

I see two input devices associated to the keyboard when I do `evemu-record`  In one of them I get events but radio-silence from the other.

I am new to FreeBSD, and at this point I am extremely curious of what is happening, maybe the keyboard is just not supported. Anyways, it would be awesome if you could guide me on how to debug and maybe even try to add support, I am also in the process of learning low level programming and playing with the drivers would not be a bad idea.

[`devinfo -rv`](https://gist.github.com/goyox86/a7c3f40b60915b32d9254dafcadad7b0)
[`pciconf -lv`](https://gist.github.com/goyox86/414296fdea540985acda8422892e8d20)
[`sudo `usbconfig`](https://gist.github.com/goyox86/249a5a57414333db41b23f7cc7a825f9)
[`dmesg`](https://gist.github.com/goyox86/28e90b66fbee2dfb8730da38254ff89c)

As I mentioned `ukbd` works very nicely with non dual keys, and it is really not of a big deal but it would be great if I could make them work like in Linux/MacOS or Windows where they are working out of the box. 

It would be great to have some pointers on how we could make it work :) 
Let me know if need anything else!

And thanks again!

wulf7, 2021-04-26:
Hi @goyox86 !

README is little bit stalled. You should add following
```
hw.usb.usbhid.enable=1
usbhid_load="YES"
```
lines to `/boot//loader.conf` to enable usbhid.ko and h* set of drivers.

goyox86, 2021-04-27:
It works!

I had to prevent my `ukbd` and `ums` modules from building in my custom kernel (`WITHOUT_MODULES=usb/ukbd usb/ums` in `/etc/make.conf`). Also removed them from being linked directly in the kernel.

Then it works like a charm! 

Thank you again for these drivers!

goyox86, 2021-04-29:
Closing this as solved!

closed 2021-04-29 by goyox86