wulf7/iichid #60: Does not compile on -CURRENT

sinetek, 2021-05-07:
Hi, my machine runs Current.

I can't compile. here is the log.

```
./hid.h:122:14: error: redefinition of typedef 'hid_test_quirk_t' is a C11 feature [-Werror,-Wtypedef-redefinition]
typedef bool hid_test_quirk_t(const struct hid_device_info *dev_info,
             ^
/usr/src/sys/dev/hid/hid.h:306:14: note: previous definition is here
typedef bool hid_test_quirk_t(const struct hid_device_info *dev_info,
             ^
In file included from iichid.c:57:
./hid.h:126:1: error: static declaration of 'hid_get_udata' follows non-static declaration
hid_get_udata(const uint8_t *buf, hid_size_t len, struct hid_location *loc)
^
/usr/src/sys/dev/hid/hid.h:325:10: note: previous declaration is here
uint32_t hid_get_udata(const uint8_t *buf, hid_size_t len,
         ^
In file included from iichid.c:57:
./hid.h:128:10: error: implicit declaration of function 'hid_get_data_unsigned' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        return (hid_get_data_unsigned(buf, len, loc));
                ^
11 errors generated.
*** Error code 1

Stop.
make: stopped in /usr/home/phil/projects/iichid
[phil@philhost:~/projects/iichid]$ 
```

wulf7, 2021-05-07:
Hi Philippe,

This version of iichid is for 12 only. 13 and 14-CURRENT contain it in base system already.

P.S. On 13+ It is split on dozen of drivers.

wulf7, 2021-05-07:
USB support is disabled in 13+ by default. You should add following lines to /boot/loader.conf to enable it:
```
hw.usb.usbhid.enable=1
usbhid_load="YES"
```

closed 2022-02-14 by wulf7