wulf7/iichid #61: Bad mapping of Xbox 360 Wired controller

heldernovais, 2021-07-11:
* I plug the controller into the computer, running FreeBSD 13.0 STABLE. dmesg properly detects a Microsoft controller.  I manually load up all the modules, just in case (xb360gp, hgame, iichid, usbus, ...).
* The XBOX button's lights flash and then turn themselves off after about 5 turns. Reconnecting it (USB 1.1? black port) will repeat this. The player1 light is meant to stay on, as it does on Windows (same port, as far as I remember), Linux (same port) and the 360 console.
* The input is tested using emulators (mednafen uses evdev, afaik).
* Face buttons, bumpers and right axis appear to work. Left axis -X doesn't seem to be registered. The d-pad doesn't work at all. Triggers aren't detected either. Tested using mednafen and nestopia.

I wanted to try experimenting with the code to fix it but I am currently unable to. Could anyone here take a look into this?

wulf7, 2021-07-11:
What evemu-record from devel/evemu port shows when you use dysfunctional controls?

heldernovais, 2021-07-12:
Thank you for your reply.

First of all:
I experimented using a PSX rom with mednafen first and I mapped all the virtual buttons to button B.
'psx.input.port1.gamepad.circle joystick 0xdcae89a8555bb8b10006000b00000000 button_8'
This is from the cfg file. I imagine that's the code for the joystick followed with button 8, supposedly B, being mapped to PSX circle.
I look at your code and it seems like button 8 is actually mapped to the start button and not B?

Strangely, I did 'kldunload xb360gp' and tried remapping. Mednafen is still reading the inputs.
I did 'kldunload iichid' and ... same result. It still reads the inputs. The cfg line even looks the same.
What could be reading my joystick instead? This is a nearly fresh install of FreeBSD 13 STABLE.

Secondly:
I ran evemu-record and the xbox controller wasn't even listed. All /dev/input/eventX map to other inputs.
I tested all of them, just in case, and input from the controller was never detected.

Thirdly: 
This is the dmesg output: 
ugen1.4: <Microsoft Corporation Controller> at usbus1
uhid1 on uhub3
uhid1: <Microsoft Corporation Controller, rev 2.00/1.14, addr 4> on usbus1

Is it being handled by another module instead? I did reload the modules.
'usbhidctl -a -f /dev/uhid1' does print the the action in the controllers, but this time, button B is mapped to button 1:
'Generic_Desktop:Game_Pad.Button:Button_2=1'

heldernovais, 2021-07-12:
Update:
doing usbhidctl with '-l' for looping shows all of the buttons being recognised and mapped correctly. But these aren't read properly by the programs I tried it on.

I forgot that I needed to press F3 before testing the axis on mednafen. It reads the triggers and analogue sticks but it still can't read the d-pad at all. I believe mednafen would rather read the d-pad as an axis as that's the way it works on Linux, at least when using <Linux/joystick.h> api instead of evdev.

Is there a way to remap the d-pad buttons to 2 xy axis?

wulf7, 2021-07-12:
> This is the dmesg output:
> ugen1.4: at usbus1
> uhid1 on uhub3
> uhid1: <Microsoft Corporation Controller, rev 2.00/1.14, addr 4> on usbus1

You did not enable usbhid. Add following lines to `/boot/loader.conf`
```
hw.usb.usbhid.enable=1
usbhid_load="YES"
```

heldernovais, 2021-07-12:
ugen1.4: <Microsoft Corporation Controller> at usbus1
usbhid0 on uhub4
usbhid0: <Microsoft Corporation Controller, rev 2.00/1.14, addr 4> on usbus1
hidbus1: <HID bus> on usbhid0
xb360gp0: <XBox 360 Gamepad> on hidbus1

Now evemu-record recognises the controller properly, but neither mednafen or nestopia will detect any input from it.

heldernovais, 2021-07-12:
I also added "xb360gp_enable=YES" and did a reboot -> same result.
I deleted the lines and now the controller is read again (by uhid, with bad mapping).

wulf7, 2021-07-13:
> Now evemu-record recognises the controller properly, but neither mednafen nor nestopia will detect any input from it.

There may be two causes:
1. You do not have enough permissions to open /dev/input/event* . See e.g. https://github.com/wulf7/iichid/issues/43
2. libsdl2 is built without evdev support

heldernovais, 2021-07-15:
* I tried running them as root, same effect.
* Nestopia doesn't use SDL, so that can't be it.

Please be more cooperative.

wulf7, 2021-07-15:
They both use SDL2.

I just checked ports tree and found out that devel/sdl20 does not support evdev yet, so you need to apply one of following patches and rebuild it:
https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=249874
https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=252371

You can test joysticks with `sdl2-jstest` utility from https://gitlab.com/sdl-jstest/sdl-jstest repo after than.