wulf7/iichid #63: c300s TERRA i2c Designware ELAN

mattdrepo, 2021-11-23:
Hi there,

Quite confused as to how to approach for this old Chromebook on FREEBSD-13.0-STABLE

I don't get any pointer when starting X.  Could someone give me a few steps as to how to fix a clean install.

My issues are that I can't get any of the guides mentioned here to work and I am confused as to how to approach things now and I dont want to try older guides due to the depreciation of moused, nor go down the route of synaptics.  I am also confused as to what to put in boot and rc. 

Sorry to be a pain, everything else about this system seems to be workable or manageable.  Thank you to anyone willing to help!

wulf7, 2021-11-23:
Please provide some info about your system:
1, bootlog. can be obtained with `dmesg` command right after boot
2. `devinfo -rv` output
3. `pciinfo -lv` ouput

mattdrepo, 2021-11-29:
Thanks for getting back to me.  I have been struggling to get the output over to you on FreeBSD.  Because I use this device quite often outdoors I have to switch between Fedora and then (when I have time) trying again with FreeBSD.  I installed pciutils to get lspci and the other commands i piped to txt files accordingly.  My issues are being (too used to hand holding in linux) a bit daunted by FreeBSD and got stuck getting these txt files off the machine for you.  Do you object to me giving you the relevant output on a linux machine.  Or is the FreeBSD output empirical? 

literally trying to get it off the laptop onto either the sdcard or a usb.

wulf7, 2021-12-06:
Ok. You may start from lspci and friends. Providing linux tools output is better than providing nothing. But native tools output is more valuable.

mattdrepo, 2021-12-09:
Apologies for the delay!

I have *finally* managed to put together a working (minus trackpad) system on this chromebook (entropy issues on all installers bar FreeBSD 13 STABLE and FreeBSD 14).  Here is the output from FreeBSD 13 STABLE

mattdrepo, 2021-12-09:
dmesg

---<<BOOT>>---
Copyright (c) 1992-2021 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 13.0-STABLE #0 stable/13-n248076-3a15ccadf87: Thu Nov 18 02:34:01 UTC 2021
    root@releng3.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
FreeBSD clang version 12.0.1 (git@github.com:llvm/llvm-project.git llvmorg-12.0.1-0-gfed41342a82f)
VT(efifb): resolution 1366x768
CPU: Intel(R) Celeron(R) CPU  N3060  @ 1.60GHz (1599.99-MHz K8-class CPU)
  Origin="GenuineIntel"  Id=0x406c4  Family=0x6  Model=0x4c  Stepping=4
  Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
  Features2=0x43d8e3bf<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,EST,TM2,SSSE3,CX16,xTPR,PDCM,SSE4.1,SSE4.2,MOVBE,POPCNT,TSCDLT,AESNI,RDRAND>
  AMD Features=0x28100800<SYSCALL,NX,RDTSCP,LM>
  AMD Features2=0x101<LAHF,Prefetch>
  Structured Extended Features=0x2282<TSCADJ,SMEP,ERMS,NFPUSG>
  Structured Extended Features3=0xc000400<MD_CLEAR,IBPB,STIBP>
  VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID
  TSC: P-state invariant, performance statistics
real memory  = 2147483648 (2048 MB)
avail memory = 1983930368 (1892 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <COREv4 COREBOOT>
WARNING: L1 data cache covers fewer APIC IDs than a core (0 < 1)
FreeBSD/SMP: Multiprocessor System Detected: 2 CPUs
FreeBSD/SMP: 1 package(s) x 2 core(s)
random: registering fast source Intel Secure Key RNG
random: fast provider: "Intel Secure Key RNG"
random: unblocking device.
ioapic0: MADT APIC ID 2 != hw id 1
ioapic0 <Version 2.0> irqs 0-114
Launching APs: 1
Timecounter "TSC" frequency 1599988320 Hz quality 1000
random: entropy device external interface
kbd1 at kbdmux0
efirtc0: <EFI Realtime Clock>
efirtc0: registered as a time-of-day clock, resolution 1.000000s
aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS>
acpi0: <COREv4 COREBOOT>
acpi0: Power Button (fixed)
ACPI Error: No handler for Region [EMEM] (0xfffff80001704a00) [EmbeddedControl] (20201113/evregion-290)
ACPI Error: Region EmbeddedControl (ID=3) has no handler (20201113/exfldio-428)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BSTA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BAT0._STA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Timecounter "HPET" frequency 14318180 Hz quality 950
Event timer "HPET" frequency 14318180 Hz quality 450
Event timer "HPET1" frequency 14318180 Hz quality 440
Event timer "HPET2" frequency 14318180 Hz quality 440
cpu0: <ACPI CPU> on acpi0
atrtc0: <AT realtime clock> port 0x70-0x77 on acpi0
atrtc0: registered as a time-of-day clock, resolution 1.000000s
Event timer "RTC" frequency 32768 Hz quality 0
attimer0: <AT timer> port 0x40-0x43,0x50-0x53 irq 0 on acpi0
Timecounter "i8254" frequency 1193182 Hz quality 0
Event timer "i8254" frequency 1193182 Hz quality 100
Timecounter "ACPI-fast" frequency 3579545 Hz quality 900
acpi_timer0: <24-bit timer at 3.579545MHz> port 0x408-0x40b on acpi0
acpi_ec0: <Embedded Controller: GPE 0x10> port 0x62,0x66 on acpi0
pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0
pcib0: could not evaluate _ADR - AE_NOT_FOUND
pci0: <ACPI PCI bus> on pcib0
vgapci0: <VGA-compatible display> port 0x1000-0x103f mem 0x90000000-0x90ffffff,0x80000000-0x8fffffff at device 2.0 on pci0
vgapci0: Boot video device
sdhci_pci0: <Intel Braswell eMMC 4.5.1 Controller> mem 0x9131d000-0x9131dfff at device 16.0 on pci0
sdhci_pci0: 1 slot(s) allocated
mmc0: <MMC/SD bus> on sdhci_pci0
sdhci_pci1: <Intel Braswell SDXC Controller> mem 0x9131e000-0x9131efff at device 18.0 on pci0
[... 68 lines cut ...]
[drm]   - kern.vt.fb.default_mode
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] Initialized i915 1.6.0 20190822 for drmn0 on minor 0
WARNING: Device "fb" is Giant locked and may be deleted before FreeBSD 14.0.
VT: Replacing driver "dummy" with new "fb".
start FB_INFO:
type=11 height=768 width=1366 depth=32
cmsize=16 size=4227072
pbase=0x80000000 vbase=0xfffff80080000000
name=drmn0 flags=0x0 stride=5504 bpp=32
cmap[0]=0 cmap[1]=7f0000 cmap[2]=7f00 cmap[3]=c4a000
end FB_INFO
drmn0: fb0: i915drmfb frame buffer device
iwm0: <Intel(R) Dual Band Wireless AC 7265> mem 0x91200000-0x91201fff at device 0.0 on pci1
iwm0: hw rev 0x210, fw ver 22.361476.0, address 7c:b0:c2:26:37:bb
wlan0: Ethernet address: 7c:b0:c2:26:37:bb
lo0: link state changed to UP
wlan0: link state changed to UP
ums0 on uhub0
ums0: <PixArt Microsoft USB Optical Mouse, class 0/0, rev 1.10/1.00, addr 2> on usbus0
ums0: 3 buttons and [XYZ] coordinates ID=0
ugen0.4: <vendor 0x8087 product 0x0a2a> at usbus0 (disconnected)
ugen0.4: <vendor 0x8087 product 0x0a2a> at usbus0
ng_ubt: HCI command 0x0c03 timed out
ng_ubt: HCI command 0x0c03 timed out

mattdrepo, 2021-12-09:
devinfo -rv

nexus0
  efirtc0
  cryptosoft0
  aesni0
  ram0
      I/O memory addresses:
          0x0-0x9ffff
          0x100000-0x1fa3cfff
          0x1fa3e000-0x1fc71fff
          0x1fc75000-0x1fc77fff
          0x1fc7b000-0x1feaefff
          0x1ffff000-0x1fffffff
          0x20200000-0x7ccddfff
  apic0
      I/O memory addresses:
          0xfec00000-0xfec0001f
          0xfee00000-0xfee0053f
  acpi0
      Interrupt request lines:
          0x9
      I/O ports:
          0x61
          0x63
          0x65
          0x67
          0x80
          0x92
          0xb2-0xb3
          0x200
          0x204
          0x400-0x47f
          0x500-0x5fe
          0x800-0x87f
          0x880-0x8ff
          0x900-0x9fe
      I/O memory addresses:
          0xe0000000-0xefffffff
          0xfea00000-0xfeafffff
          0xfeb00000-0xfebfffff
          0xfed01000-0xfed013ff
          0xfed03000-0xfed033ff
          0xfed06000-0xfed067ff
          0xfed08000-0xfed09fff
          0xfed1c000-0xfed1c3ff
    pcib0 pnpinfo _HID=PNP0A08 _UID=0 _CID=PNP0A03 at handle=\_SB_.PCI0
        I/O ports:
            0xcf8-0xcff
      pci0
          PCI domain 0 bus numbers:
              0
        hostb0 pnpinfo vendor=0x8086 device=0x2280 subvendor=0x8086 subdevice=0x7270 class=0x060000 at slot=0 function=0 dbsf=pci0:0:0:0
        vgapci0 pnpinfo vendor=0x8086 device=0x22b1 subvendor=0x8086 subdevice=0x7270 class=0x030000 at slot=2 function=0 dbsf=pci0:0:2:0 handle=\_SB_.PCI0.GFX0
            Interrupt request lines:
                0x76
            I/O ports:
                0x1000-0x103f
            I/O memory addresses:
                0x80000000-0x8fffffff
                0x90000000-0x90ffffff
          drm0
            drm7
              i2c0
            drm8
              i2c1
            drm9
            drm10
              i2c2
            drm11
[... 190 lines cut ...]
            0xfed90000-0xfed97fff
    unknown pnpinfo _HID=INT33FF _UID=4 _CID=INT33FF at handle=\_SB_.GPSE
        I/O memory addresses:
            0xfed98000-0xfed9ffff
    unknown pnpinfo _HID=INT3400 _UID=0 _CID=none at handle=\_SB_.DPTF
    unknown pnpinfo _HID=INT3403 _UID=1 _CID=none at handle=\_SB_.DPTF.TSR0
    unknown pnpinfo _HID=INT3403 _UID=2 _CID=none at handle=\_SB_.DPTF.TSR1
    unknown pnpinfo _HID=INT3403 _UID=3 _CID=none at handle=\_SB_.DPTF.TSR2
    unknown pnpinfo _HID=INT3403 _UID=0 _CID=none at handle=\_SB_.DPTF.TCHG
    cpu0 pnpinfo _HID=none _UID=0 _CID=none at handle=\_SB_.CP00
      est0
      p4tcc0
      acpi_perf0
      cpufreq0
    cpu1 pnpinfo _HID=none _UID=0 _CID=none at handle=\_SB_.CP01
      est1
      p4tcc1
      acpi_perf1
      cpufreq1
    unknown pnpinfo _HID=BOOT0000 _UID=0 _CID=none at handle=\CTBL
        I/O memory addresses:
            0x7cd13000-0x7cd1afff
    acpi_timer0 pnpinfo unknown
        ACPI I/O ports:
            0x408-0x40b

mattdrepo, 2021-12-09:
pciconf -l (pciinfo -lv wasn't found and couldnt find a pkg for it??)

hostb0@pci0:0:0:0:	class=0x060000 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2280 subvendor=0x8086 subdevice=0x7270
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x35 hdr=0x00 vendor=0x8086 device=0x22b1 subvendor=0x8086 subdevice=0x7270
none0@pci0:0:11:0:	class=0x118000 rev=0x35 hdr=0x00 vendor=0x8086 device=0x22dc subvendor=0x8086 subdevice=0x7270
sdhci_pci0@pci0:0:16:0:	class=0x080501 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2294 subvendor=0x8086 subdevice=0x7270
sdhci_pci1@pci0:0:18:0:	class=0x080501 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2296 subvendor=0x8086 subdevice=0x7270
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x35 hdr=0x00 vendor=0x8086 device=0x22b5 subvendor=0x8086 subdevice=0x7270
hdac0@pci0:0:27:0:	class=0x040300 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2284 subvendor=0x8086 subdevice=0x7270
pcib1@pci0:0:28:0:	class=0x060400 rev=0x35 hdr=0x01 vendor=0x8086 device=0x22c8 subvendor=0x8086 subdevice=0x7270
pcib2@pci0:0:28:2:	class=0x060400 rev=0x35 hdr=0x01 vendor=0x8086 device=0x22cc subvendor=0x8086 subdevice=0x7270
isab0@pci0:0:31:0:	class=0x060100 rev=0x35 hdr=0x00 vendor=0x8086 device=0x229c subvendor=0x8086 subdevice=0x7270
iwm0@pci0:2:0:0:	class=0x028000 rev=0x59 hdr=0x00 vendor=0x8086 device=0x095a subvendor=0x8086 subdevice=0x5010

wulf7, 2021-12-09:
Could you apply following patch and than rebuild and test your kernel ?
```
--- sys/dev/ichiic/ig4_acpi.c
+++ sys/dev/ichiic/ig4_acpi.c
@@ -193,3 +193,4 @@ static driver_t ig4iic_acpi_driver = {
 DRIVER_MODULE_ORDERED(ig4iic, acpi, ig4iic_acpi_driver, ig4iic_devclass, 0, 0,
     SI_ORDER_ANY);
 MODULE_DEPEND(ig4iic, acpi, 1, 1, 1);
+ACPI_PNP_INFO(ig4iic_ids);
```

mattdrepo, 2021-12-14:
Hello again!

Apologies for the delay! (this machine takes a LONG time to build from source!!!)

rm -rf /usr/src/* /usr/src/.[^.]*
git clone https://git.freebsd.org/src.git --branch stable/13 /usr/src

#then applied your patch

#building from source
cd /usr/src
make buildworld
make buildkernel
make installkernel
make installworld

could you tell me now what you require now that the system is patched?

Thank you for your time!

wulf7, 2021-12-15:
I suspect that after reboot your system will automatically load required drivers.
If not, try to execute `kldload ig4.ko`

mattdrepo, 2021-12-15:
Hello again!

I have added:

ig4_load="YES"
iicbus_load="YES"
iichid_load="YES"

to /boot/loader.conf

and have polled devices via
libinput list-devices

I only have this listed under mouse criteria

Device:           System mouse
Kernel:           /dev/input/event0
Group:            1
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

other than my usb mouse connected.

could you give me pointers (no pun intended!) as to how to proceed?

Thank you for your "input" :-)

mattdrepo, 2021-12-15:
don't think (going by other peoples output on systems with a mousepad) I am seeing anything that suggests a mousepad on this system.  I also find all the guides to getting mousepad/touchpad support on FreeBSD confusing (mainly due to the differing methods per FreeBSD version and, how many methods a mousepad is connected to a system).  Hope you can help!

Considering this is a chromebook, I have discovered (when using this machine on Linux - galliumos patches specifically) it reads as elan/elantech

dmesg (after patch)

---<<BOOT>>---
Copyright (c) 1992-2021 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 13.0-STABLE #0 stable/13-n248516-7572ddd1671-dirty: Tue Dec 14 20:18:02 GMT 2021
    matt@netbook:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
FreeBSD clang version 13.0.0 (git@github.com:llvm/llvm-project.git llvmorg-13.0.0-0-gd7b669b3a303)
VT(efifb): resolution 1366x768
CPU: Intel(R) Celeron(R) CPU  N3060  @ 1.60GHz (1599.99-MHz K8-class CPU)
  Origin="GenuineIntel"  Id=0x406c4  Family=0x6  Model=0x4c  Stepping=4
  Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
  Features2=0x43d8e3bf<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,EST,TM2,SSSE3,CX16,xTPR,PDCM,SSE4.1,SSE4.2,MOVBE,POPCNT,TSCDLT,AESNI,RDRAND>
  AMD Features=0x28100800<SYSCALL,NX,RDTSCP,LM>
  AMD Features2=0x101<LAHF,Prefetch>
  Structured Extended Features=0x2282<TSCADJ,SMEP,ERMS,NFPUSG>
  Structured Extended Features3=0xc000400<MD_CLEAR,IBPB,STIBP>
  VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID
  TSC: P-state invariant, performance statistics
real memory  = 2147483648 (2048 MB)
avail memory = 1983930368 (1892 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <COREv4 COREBOOT>
WARNING: L1 data cache covers fewer APIC IDs than a core (0 < 1)
FreeBSD/SMP: Multiprocessor System Detected: 2 CPUs
FreeBSD/SMP: 1 package(s) x 2 core(s)
random: registering fast source Intel Secure Key RNG
random: fast provider: "Intel Secure Key RNG"
random: unblocking device.
ioapic0: MADT APIC ID 2 != hw id 1
ioapic0 <Version 2.0> irqs 0-114
Launching APs: 1
Timecounter "TSC" frequency 1599987860 Hz quality 1000
random: entropy device external interface
kbd1 at kbdmux0
efirtc0: <EFI Realtime Clock>
efirtc0: registered as a time-of-day clock, resolution 1.000000s
aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS>
acpi0: <COREv4 COREBOOT>
acpi0: Power Button (fixed)
ACPI Error: No handler for Region [EMEM] (0xfffff80001706a00) [EmbeddedControl] (20201113/evregion-290)
ACPI Error: Region EmbeddedControl (ID=3) has no handler (20201113/exfldio-428)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BSTA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BAT0._STA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Timecounter "HPET" frequency 14318180 Hz quality 950
Event timer "HPET" frequency 14318180 Hz quality 450
Event timer "HPET1" frequency 14318180 Hz quality 440
Event timer "HPET2" frequency 14318180 Hz quality 440
cpu0: <ACPI CPU> on acpi0
atrtc0: <AT realtime clock> port 0x70-0x77 on acpi0
atrtc0: registered as a time-of-day clock, resolution 1.000000s
Event timer "RTC" frequency 32768 Hz quality 0
attimer0: <AT timer> port 0x40-0x43,0x50-0x53 irq 0 on acpi0
Timecounter "i8254" frequency 1193182 Hz quality 0
Event timer "i8254" frequency 1193182 Hz quality 100
Timecounter "ACPI-fast" frequency 3579545 Hz quality 900
acpi_timer0: <24-bit timer at 3.579545MHz> port 0x408-0x40b on acpi0
acpi_ec0: <Embedded Controller: GPE 0x10> port 0x62,0x66 on acpi0
pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0
pcib0: could not evaluate _ADR - AE_NOT_FOUND
pci0: <ACPI PCI bus> on pcib0
vgapci0: <VGA-compatible display> port 0x1000-0x103f mem 0x90000000-0x90ffffff,0x80000000-0x8fffffff at device 2.0 on pci0
vgapci0: Boot video device
[... 79 lines cut ...]
type=11 height=768 width=1366 depth=32
cmsize=16 size=4227072
pbase=0x80000000 vbase=0xfffff80080000000
name=drmn0 flags=0x0 stride=5504 bpp=32
cmap[0]=0 cmap[1]=7f0000 cmap[2]=7f00 cmap[3]=c4a000
end FB_INFO
drmn0: fb0: i915drmfb frame buffer device
iwm0: <Intel(R) Dual Band Wireless AC 7265> mem 0x91200000-0x91201fff at device 0.0 on pci1
iwm0: hw rev 0x210, fw ver 22.361476.0, address 7c:b0:c2:26:37:bb
ig4iic0: <Designware I2C Controller> iomem 0x91323000-0x91323fff irq 36 on acpi0
iicbus0: <Philips I2C bus (ACPI-hinted)> on ig4iic0
iicbus0: <unknown card> at addr 0x1a
ig4iic1: <Designware I2C Controller> iomem 0x91325000-0x91325fff irq 37 on acpi0
iicbus1: <Philips I2C bus (ACPI-hinted)> on ig4iic1
iicbus1: <unknown card> at addr 0x15
wlan0: Ethernet address: 7c:b0:c2:26:37:bb
lo0: link state changed to UP
wlan0: link state changed to UP
ums0 on uhub0
ums0: <PixArt Microsoft USB Optical Mouse, class 0/0, rev 1.10/1.00, addr 1> on usbus0
ums0: 3 buttons and [XYZ] coordinates ID=0
ugen0.3: <vendor 0x8087 product 0x0a2a> at usbus0 (disconnected)
ugen0.3: <vendor 0x8087 product 0x0a2a> at usbus0
ng_ubt: HCI command 0x0c03 timed out
ng_ubt: HCI command 0x0c03 timed out

mattdrepo, 2021-12-15:
pciconf

hostb0@pci0:0:0:0:	class=0x060000 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2280 subvendor=0x8086 subdevice=0x7270
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x35 hdr=0x00 vendor=0x8086 device=0x22b1 subvendor=0x8086 subdevice=0x7270
none0@pci0:0:11:0:	class=0x118000 rev=0x35 hdr=0x00 vendor=0x8086 device=0x22dc subvendor=0x8086 subdevice=0x7270
sdhci_pci0@pci0:0:16:0:	class=0x080501 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2294 subvendor=0x8086 subdevice=0x7270
sdhci_pci1@pci0:0:18:0:	class=0x080501 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2296 subvendor=0x8086 subdevice=0x7270
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x35 hdr=0x00 vendor=0x8086 device=0x22b5 subvendor=0x8086 subdevice=0x7270
hdac0@pci0:0:27:0:	class=0x040300 rev=0x35 hdr=0x00 vendor=0x8086 device=0x2284 subvendor=0x8086 subdevice=0x7270
pcib1@pci0:0:28:0:	class=0x060400 rev=0x35 hdr=0x01 vendor=0x8086 device=0x22c8 subvendor=0x8086 subdevice=0x7270
pcib2@pci0:0:28:2:	class=0x060400 rev=0x35 hdr=0x01 vendor=0x8086 device=0x22cc subvendor=0x8086 subdevice=0x7270
isab0@pci0:0:31:0:	class=0x060100 rev=0x35 hdr=0x00 vendor=0x8086 device=0x229c subvendor=0x8086 subdevice=0x7270
iwm0@pci0:2:0:0:	class=0x028000 rev=0x59 hdr=0x00 vendor=0x8086 device=0x095a subvendor=0x8086 subdevice=0x5010

mattdrepo, 2021-12-15:
devinfo:

nexus0
  efirtc0
  cryptosoft0
  aesni0
  ram0
      I/O memory addresses:
          0x0-0x9ffff
          0x100000-0x1fa3cfff
          0x1fa3e000-0x1fc71fff
          0x1fc75000-0x1fc77fff
          0x1fc7b000-0x1feaefff
          0x1ffff000-0x1fffffff
          0x20200000-0x7ccddfff
  apic0
      I/O memory addresses:
          0xfec00000-0xfec0001f
          0xfee00000-0xfee0053f
  acpi0
      Interrupt request lines:
          0x9
      I/O ports:
          0x61
          0x63
          0x65
          0x67
          0x80
          0x92
          0xb2-0xb3
          0x200
          0x204
          0x400-0x47f
          0x500-0x5fe
          0x800-0x87f
          0x880-0x8ff
          0x900-0x9fe
      I/O memory addresses:
          0xe0000000-0xefffffff
          0xfea00000-0xfeafffff
          0xfeb00000-0xfebfffff
          0xfed01000-0xfed013ff
          0xfed03000-0xfed033ff
          0xfed06000-0xfed067ff
          0xfed08000-0xfed09fff
          0xfed1c000-0xfed1c3ff
    pcib0 pnpinfo _HID=PNP0A08 _UID=0 _CID=PNP0A03 at handle=\_SB_.PCI0
        I/O ports:
            0xcf8-0xcff
      pci0
          PCI domain 0 bus numbers:
              0
        hostb0 pnpinfo vendor=0x8086 device=0x2280 subvendor=0x8086 subdevice=0x7270 class=0x060000 at slot=0 function=0 dbsf=pci0:0:0:0
        vgapci0 pnpinfo vendor=0x8086 device=0x22b1 subvendor=0x8086 subdevice=0x7270 class=0x030000 at slot=2 function=0 dbsf=pci0:0:2:0 handle=\_SB_.PCI0.GFX0
            Interrupt request lines:
                0x76
            I/O ports:
                0x1000-0x103f
            I/O memory addresses:
                0x80000000-0x8fffffff
                0x90000000-0x90ffffff
          drm0
            drm7
              i2c0
            drm8
              i2c1
            drm9
            drm10
              i2c2
            drm11
[... 489 lines cut ...]
            0xfed90000-0xfed97fff
    unknown pnpinfo _HID=INT33FF _UID=4 _CID=INT33FF at handle=\_SB_.GPSE
        I/O memory addresses:
            0xfed98000-0xfed9ffff
    unknown pnpinfo _HID=INT3400 _UID=0 _CID=none at handle=\_SB_.DPTF
    unknown pnpinfo _HID=INT3403 _UID=1 _CID=none at handle=\_SB_.DPTF.TSR0
    unknown pnpinfo _HID=INT3403 _UID=2 _CID=none at handle=\_SB_.DPTF.TSR1
    unknown pnpinfo _HID=INT3403 _UID=3 _CID=none at handle=\_SB_.DPTF.TSR2
    unknown pnpinfo _HID=INT3403 _UID=0 _CID=none at handle=\_SB_.DPTF.TCHG
    cpu0 pnpinfo _HID=none _UID=0 _CID=none at handle=\_SB_.CP00
      est0
      p4tcc0
      acpi_perf0
      cpufreq0
    cpu1 pnpinfo _HID=none _UID=0 _CID=none at handle=\_SB_.CP01
      est1
      p4tcc1
      acpi_perf1
      cpufreq1
    unknown pnpinfo _HID=BOOT0000 _UID=0 _CID=none at handle=\CTBL
        I/O memory addresses:
            0x7cd13000-0x7cd1afff
    acpi_timer0 pnpinfo unknown
        ACPI I/O ports:
            0x408-0x40b

wulf7, 2021-12-16:
```
ig4iic1 pnpinfo _HID=808622C1 _UID=6 CID=none at handle=_SB.PCI0.I2C6
Interrupt request lines:
0x25
I/O memory addresses:
0x91325000-0x91325fff
iicbus1
unknown pnpinfo _HID=ELAN0000 _UID=3 CID=none at addr=0x15 handle=_SB.PCI0.I2C6.ETPA
```
Last line is your touchpad. Unfortunately it does not have proper _CID (_CID=PNP0C50) to be handled by iichid and uses unsupported proprietary protocol. There are some chances that it has HID-compatible fallback mode which you may try to activate with following patch:
```
--- sys/dev/iicbus/iichid.c
+++ sys/dev/iicbus/iichid.c
@@ -133,6 +133,7 @@ struct i2c_hid_desc {
 static char *iichid_ids[] = {
 	"PNP0C50",
 	"ACPI0C50",
+	"ELAN0000",
 	NULL
 };
 
```
Add `hw.iichid.debug=1` to `/boot/loader.conf` to get verbose iichid boot log if you are going to test the patch.

P.S. It is not necessary to rebuild world to build new kernel. Just use `make kernel` to build and install only a kernel.

mattdrepo, 2021-12-17:
Hi there!

Thanks for the input

It didnt work but I thought I might share with you some possible input.

https://duckduckgo.com/?q=ELAN0000+galliumos&t=newext&atb=v291-1&ia=web

seems this laptop has space to have both touchscreen and mouse on the i2c, under linux very early on in the galliumos development blacklisting one input allowed access to the other.  I will share the output of dmesg etc before reverting to the linux system i use as a fallback.  Perhaps you might have an insight as to what to blacklist, or a means to experiment?  If not, I totally understand it might be too much to ask. In which case I will revert to using linux and ssh to other FreeBSD boxes from this.

Thanks for helping so much!

mattdrepo, 2021-12-17:
dmsg

---<<BOOT>>---
Copyright (c) 1992-2021 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 13.0-STABLE #0 stable/13-n248076-3a15ccadf87: Thu Nov 18 02:34:01 UTC 2021
    root@releng3.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
FreeBSD clang version 12.0.1 (git@github.com:llvm/llvm-project.git llvmorg-12.0.1-0-gfed41342a82f)
VT(efifb): resolution 1366x768
CPU: Intel(R) Celeron(R) CPU  N3060  @ 1.60GHz (1599.99-MHz K8-class CPU)
  Origin="GenuineIntel"  Id=0x406c4  Family=0x6  Model=0x4c  Stepping=4
  Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
  Features2=0x43d8e3bf<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,EST,TM2,SSSE3,CX16,xTPR,PDCM,SSE4.1,SSE4.2,MOVBE,POPCNT,TSCDLT,AESNI,RDRAND>
  AMD Features=0x28100800<SYSCALL,NX,RDTSCP,LM>
  AMD Features2=0x101<LAHF,Prefetch>
  Structured Extended Features=0x2282<TSCADJ,SMEP,ERMS,NFPUSG>
  Structured Extended Features3=0xc000400<MD_CLEAR,IBPB,STIBP>
  VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID
  TSC: P-state invariant, performance statistics
real memory  = 2147483648 (2048 MB)
avail memory = 1983930368 (1892 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <COREv4 COREBOOT>
WARNING: L1 data cache covers fewer APIC IDs than a core (0 < 1)
FreeBSD/SMP: Multiprocessor System Detected: 2 CPUs
FreeBSD/SMP: 1 package(s) x 2 core(s)
random: registering fast source Intel Secure Key RNG
random: fast provider: "Intel Secure Key RNG"
random: unblocking device.
ioapic0: MADT APIC ID 2 != hw id 1
ioapic0 <Version 2.0> irqs 0-114
Launching APs: 1
Timecounter "TSC" frequency 1599988000 Hz quality 1000
random: entropy device external interface
kbd1 at kbdmux0
efirtc0: <EFI Realtime Clock>
efirtc0: registered as a time-of-day clock, resolution 1.000000s
aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS>
acpi0: <COREv4 COREBOOT>
acpi0: Power Button (fixed)
ACPI Error: No handler for Region [EMEM] (0xfffff800016d9780) [EmbeddedControl] (20201113/evregion-290)
ACPI Error: Region EmbeddedControl (ID=3) has no handler (20201113/exfldio-428)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BSTA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BAT0._STA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Timecounter "HPET" frequency 14318180 Hz quality 950
Event timer "HPET" frequency 14318180 Hz quality 450
Event timer "HPET1" frequency 14318180 Hz quality 440
Event timer "HPET2" frequency 14318180 Hz quality 440
cpu0: <ACPI CPU> on acpi0
atrtc0: <AT realtime clock> port 0x70-0x77 on acpi0
atrtc0: registered as a time-of-day clock, resolution 1.000000s
Event timer "RTC" frequency 32768 Hz quality 0
attimer0: <AT timer> port 0x40-0x43,0x50-0x53 irq 0 on acpi0
Timecounter "i8254" frequency 1193182 Hz quality 0
Event timer "i8254" frequency 1193182 Hz quality 100
Timecounter "ACPI-fast" frequency 3579545 Hz quality 900
acpi_timer0: <24-bit timer at 3.579545MHz> port 0x408-0x40b on acpi0
acpi_ec0: <Embedded Controller: GPE 0x10> port 0x62,0x66 on acpi0
pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0
pcib0: could not evaluate _ADR - AE_NOT_FOUND
pci0: <ACPI PCI bus> on pcib0
vgapci0: <VGA-compatible display> port 0x1000-0x103f mem 0x90000000-0x90ffffff,0x80000000-0x8fffffff at device 2.0 on pci0
vgapci0: Boot video device
sdhci_pci0: <Intel Braswell eMMC 4.5.1 Controller> mem 0x9131d000-0x9131dfff at device 16.0 on pci0
sdhci_pci0: 1 slot(s) allocated
mmc0: <MMC/SD bus> on sdhci_pci0
sdhci_pci1: <Intel Braswell SDXC Controller> mem 0x9131e000-0x9131efff at device 18.0 on pci0
[... 412 lines cut ...]
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-2: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-2
[drm]   - kern.vt.fb.default_mode
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] Initialized i915 1.6.0 20190822 for drmn0 on minor 0
WARNING: Device "fb" is Giant locked and may be deleted before FreeBSD 14.0.
VT: Replacing driver "dummy" with new "fb".
start FB_INFO:
type=11 height=768 width=1366 depth=32
cmsize=16 size=4227072
pbase=0x80000000 vbase=0xfffff80080000000
name=drmn0 flags=0x0 stride=5504 bpp=32
cmap[0]=0 cmap[1]=7f0000 cmap[2]=7f00 cmap[3]=c4a000
end FB_INFO
drmn0: fb0: i915drmfb frame buffer device
iwm0: <Intel(R) Dual Band Wireless AC 7265> mem 0x91200000-0x91201fff at device 0.0 on pci1
iwm0: hw rev 0x210, fw ver 22.361476.0, address 7c:b0:c2:26:37:bb
wlan0: Ethernet address: 7c:b0:c2:26:37:bb
lo0: link state changed to UP
wlan0: link state changed to UP
ugen0.2: <vendor 0x8087 product 0x0a2a> at usbus0 (disconnected)
ugen0.2: <vendor 0x8087 product 0x0a2a> at usbus0
ng_ubt: HCI command 0x0c03 timed out
ng_ubt: HCI command 0x0c03 timed out

mattdrepo, 2021-12-17:
devinfo

nexus0
  efirtc0
  cryptosoft0
  aesni0
  ram0
  apic0
  acpi0
    pcib0
      pci0
        hostb0
        vgapci0
          drmn0
        sdhci_pci0
          mmc0
            mmcsd0
        sdhci_pci1
          mmc1
            mmcsd1
        xhci0
          usbus0
            uhub0
        hdac0
          hdacc0
            hdaa0
        pcib1
        pcib2
          pci1
            iwm0
        isab0
          isa0
            uart0
    acpi_sysresource0
    pci_link0
    pci_link1
    pci_link2
    pci_link3
    pci_link4
    pci_link5
    pci_link6
    pci_link7
    acpi_ec0
    acpi_lid0
    acpi_acad0
    battery0
    atdma0
    hpet0
    acpi_sysresource1
    atrtc0
    attimer0
    acpi_sysresource2
    acpi_sysresource3
    atkbdc0
      atkbd0
    ig4iic0
      iicbus0
    ig4iic1
      iicbus1
    cpu0
      est0
      acpi_perf0
      cpufreq0
    cpu1
      est1
      acpi_perf1
      cpufreq1
    acpi_timer0

wulf7, 2021-12-18:
This chromebook misses some ACPI bits
Try this patch on top of previous two:
```
--- sys/dev/iicbus/iichid.c
+++ sys/dev/iicbus/iichid.c
@@ -1014,7 +1015,7 @@ iichid_probe(device_t dev)
 		return (ENXIO);
 
 	if (ACPI_FAILURE(iichid_get_config_reg(handle, &config_reg)))
-		return (ENXIO);
+		config_reg = 0x0001;
 
 	DPRINTF(sc, "  IICbus addr       : 0x%02X\n", sc->addr >> 1);
 	DPRINTF(sc, "  HID descriptor reg: 0x%02X\n", config_reg);
```
pciconf and devinfo output is not needed anymore. The only thing required is console output of iichid.ko load which can be obtained with e.g. `kldunload iichid.ko` `kldload iichid.ko` sequence of command. Actualy devd should execute kldload iichid.ko` automatically.

mattdrepo, 2021-12-18:
looks like the patches work to some extent :)

dmesg reports something mouse like:

Copyright (c) 1992-2021 The FreeBSD Project.
Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
	The Regents of the University of California. All rights reserved.
FreeBSD is a registered trademark of The FreeBSD Foundation.
FreeBSD 13.0-STABLE #0 stable/13-n248614-56eeb84f109-dirty: Sat Dec 18 14:17:52 +03 2021
    matt@netbook:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
FreeBSD clang version 12.0.1 (git@github.com:llvm/llvm-project.git llvmorg-12.0.1-0-gfed41342a82f)
VT(efifb): resolution 1366x768
CPU: Intel(R) Celeron(R) CPU  N3060  @ 1.60GHz (1599.99-MHz K8-class CPU)
  Origin="GenuineIntel"  Id=0x406c4  Family=0x6  Model=0x4c  Stepping=4
  Features=0xbfebfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,DTS,ACPI,MMX,FXSR,SSE,SSE2,SS,HTT,TM,PBE>
  Features2=0x43d8e3bf<SSE3,PCLMULQDQ,DTES64,MON,DS_CPL,VMX,EST,TM2,SSSE3,CX16,xTPR,PDCM,SSE4.1,SSE4.2,MOVBE,POPCNT,TSCDLT,AESNI,RDRAND>
  AMD Features=0x28100800<SYSCALL,NX,RDTSCP,LM>
  AMD Features2=0x101<LAHF,Prefetch>
  Structured Extended Features=0x2282<TSCADJ,SMEP,ERMS,NFPUSG>
  Structured Extended Features3=0xc000400<MD_CLEAR,IBPB,STIBP>
  VT-x: PAT,HLT,MTF,PAUSE,EPT,UG,VPID
  TSC: P-state invariant, performance statistics
real memory  = 2147483648 (2048 MB)
avail memory = 1983934464 (1892 MB)
Event timer "LAPIC" quality 600
ACPI APIC Table: <COREv4 COREBOOT>
WARNING: L1 data cache covers fewer APIC IDs than a core (0 < 1)
FreeBSD/SMP: Multiprocessor System Detected: 2 CPUs
FreeBSD/SMP: 1 package(s) x 2 core(s)
random: registering fast source Intel Secure Key RNG
random: fast provider: "Intel Secure Key RNG"
random: unblocking device.
ioapic0: MADT APIC ID 2 != hw id 1
ioapic0 <Version 2.0> irqs 0-114
Launching APs: 1
Timecounter "TSC" frequency 1599987860 Hz quality 1000
random: entropy device external interface
kbd1 at kbdmux0
efirtc0: <EFI Realtime Clock>
efirtc0: registered as a time-of-day clock, resolution 1.000000s
aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS>
acpi0: <COREv4 COREBOOT>
acpi0: Power Button (fixed)
ACPI Error: No handler for Region [EMEM] (0xfffff800016d5e00) [EmbeddedControl] (20201113/evregion-290)
ACPI Error: Region EmbeddedControl (ID=3) has no handler (20201113/exfldio-428)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BSTA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
ACPI Error: Aborting method \134_SB.PCI0.LPCB.EC0.BAT0._STA due to previous error (AE_NOT_EXIST) (20201113/psparse-689)
hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Timecounter "HPET" frequency 14318180 Hz quality 950
Event timer "HPET" frequency 14318180 Hz quality 450
Event timer "HPET1" frequency 14318180 Hz quality 440
Event timer "HPET2" frequency 14318180 Hz quality 440
cpu0: <ACPI CPU> on acpi0
atrtc0: <AT realtime clock> port 0x70-0x77 on acpi0
atrtc0: registered as a time-of-day clock, resolution 1.000000s
Event timer "RTC" frequency 32768 Hz quality 0
attimer0: <AT timer> port 0x40-0x43,0x50-0x53 irq 0 on acpi0
Timecounter "i8254" frequency 1193182 Hz quality 0
Event timer "i8254" frequency 1193182 Hz quality 100
Timecounter "ACPI-fast" frequency 3579545 Hz quality 900
acpi_timer0: <24-bit timer at 3.579545MHz> port 0x408-0x40b on acpi0
acpi_ec0: <Embedded Controller: GPE 0x10> port 0x62,0x66 on acpi0
pcib0: <ACPI Host-PCI bridge> port 0xcf8-0xcff on acpi0
pcib0: could not evaluate _ADR - AE_NOT_FOUND
pci0: <ACPI PCI bus> on pcib0
vgapci0: <VGA-compatible display> port 0x1000-0x103f mem 0x90000000-0x90ffffff,0x80000000-0x8fffffff at device 2.0 on pci0
vgapci0: Boot video device
sdhci_pci0: <Intel Braswell eMMC 4.5.1 Controller> mem 0x9131d000-0x9131dfff at device 16.0 on pci0
sdhci_pci0: 1 slot(s) allocated
mmc0: <MMC/SD bus> on sdhci_pci0
[... 194 lines cut ...]
Seat:             seat0, default
Capabilities:     pointer 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   button
Click methods:    none
Disable-w-typing: n/a
Accel profiles:   flat *adaptive
Rotation:         n/a

tried to copy your template
Section "InputDevice"
	Identifier "HID device"
	Driver "libinput"
	Option "Device" "/dev/input/event6"
	Option "AutoServerLayout" "true"
EndSection
replacing event6 for event4 - but no dice, what should i call the file and am I using the right context?

thank you for the help, this looks very promising now

wulf7, 2021-12-20:
> tried to copy your template

Usually you do not need any changes to xorg configuration. If you do not have any input, start with `evemu-record` utility from `devel/evemu` port. If it detects events on touch being attached to proper /dev/input/eventX ?
If no, put `hw.iichid.debug=1` line in to `/boot/loader.conf` and reboot your laptop. What is iichid0 debug output during boot?
33 buttons is way too much for touchpad, so I expect IO errors.

mattdrepo, 2021-12-20:
Can I ask a quick question so that I don't slip up.

What method of Xorg should I go for? Seems some posts about the use of iichid have mentioned using later versions or from source (to get latest) I have been installing from PKG all this time. And using the default PKG settings.

I will be installing FreeBSD 13 STABLE, building from source the kernel up to the point where your new comments will be tested.

wulf7, 2021-12-20:
You cannot install iichid from PKG on 13-any. iichid is bundled with base system starting from 13.0.
iichid in 13-STABLE has not diverged from14-CURRENT yet. So most probably you have the latest version of it if you run recent 13-STABLE

mattdrepo, 2021-12-20:
Sorry I was referring to Xorg, iichid I will be using via the means in boot/loader.conf and patching the kernel to reveal more to the system as per your instruction. My question relates to Xorg and whether I need to use a specific version just in case that is the issue.

https://larastechlab.blogspot.com/2020/01/i2c-touchpads-on-freebsd-130-current.html

commentary here refers to use of a later Xorg. Hence wondering whether stable's xorg from pkg might need to be omitted for another?

wulf7, 2021-12-20:
This article is obsolete. Modern xorg should autodetect I2C touchpads out of box.

mattdrepo, 2021-12-20:
thanks, just checking as you will understand that it can be a slow process building from source. Will report back my findings later on.

mattdrepo, 2021-12-20:
Hello again!

Here is the output of dmesg omitting the other crud and focusing on your information around iichid:

ig4iic0: <Designware I2C Controller> iomem 0x91323000-0x91323fff irq 36 on acpi0
iicbus0: <Philips I2C bus (ACPI-hinted)> on ig4iic0
iicbus0: <unknown card> at addr 0x1a
ig4iic1: <Designware I2C Controller> iomem 0x91325000-0x91325fff irq 37 on acpi0
iicbus1: <Philips I2C bus (ACPI-hinted)> on ig4iic1
iichid_get_config_reg: error evaluating _DSM
iichid0: <ELAN0000:03 04F3:0400 I2C HID device> at addr 0x15 on iicbus1
iichid0: Interrupt setup failed. Fallback to sampling
hidbus0: <HID bus> on iichid0

mattdrepo, 2021-12-20:
i used the evemu-record and piped to > event.txt as well as running it and seeing if the terminal reported anything. I think it stalled as dmesg also implies the same when looking at iichid in debug mode. Do you think this has anything to do with what i mentioned previously from the linux posts and the suggestion that the controller handles both touch screen and touch pad?  This laptop has only touchpad, but some came with touchscreens and i think that *might* be an issue perhaps?  (but then this could be a shot in the dark as it was  some of the only info i did bring up) thank you again for any suggestions.

# EVEMU 1.3
# Kernel: 13.0-STABLE
# Input device name: "ELAN0000:03 04F3:0400 Mouse"
# Input device ID: bus 0x18 vendor 0x4f3 product 0x400 version 0x01
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 272 (BTN_LEFT)
#     Event code 273 (BTN_RIGHT)
#   Event type 2 (EV_REL)
#     Event code 0 (REL_X)
#     Event code 1 (REL_Y)
#     Event code 6 (REL_HWHEEL)
#     Event code 8 (REL_WHEEL)
# Properties:
#   Property  type 0 (INPUT_PROP_POINTER)
N: ELAN0000:03 04F3:0400 Mouse
I: 0018 04f3 0400 0001
P: 01 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 03 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 43 01 00 00 00 00 00 00
B: 03 00 00 00 00 00 00 00 00
B: 04 00 00 00 00 00 00 00 00
B: 05 00 00 00 00 00 00 00 00
B: 11 00 00 00 00 00 00 00 00
B: 12 00 00 00 00 00 00 00 00
B: 14 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
################################
#      Waiting for events      #
################################

mattdrepo, 2021-12-21:
have since found the "patch" for chromiumos for this model "TERRA" and its Xorg configuration if this helps?
https://chromium.googlesource.com/chromiumos/platform/xorg-conf/+refs

wulf7, 2021-12-21:
> if this helps?

No. evemu-record must start printing of events on touch firstly.

> the suggestion that the controller handles both touch screen and touch pad?

I don`t think that presence or absence of one device can affect another one

> Here is the output of dmesg omitting the other crud and focusing on your information around iichid:

I forgot that you should add `options IICHID_DEBUG` line to your kernel config too to get debug output.
Please do following steps:

1. Rebuild kernel with aforementioned option enable
2. Disable xorg`s autorun.
3. Add `hw.iichid.debug=5` line in to `/boot/loader.conf`. Note digit 5 instead of 1
4. Reboot in to console and attach evemu-record to touchpad device node
5. Touch surface for brief period of time and than stop evemu-record with CTRL-C
6. Post part of dmesg starting from first `ig4iic0:` line here.

mattdrepo, 2021-12-21:
do you mean like so?
make buildkernel options IICHID_DEBUG

Disable xorg`s autorun - I use startx to get into i3, so I should be fine running from console ?

will update later!

wulf7, 2021-12-21:
Generic kernel config lies here: `/usr/src/sys/amd64/conf/GENERIC`. Put `options IICHID_DEBUG` in to it and do `make -jX kernel` after than. Where X is a number of CPU cores on laptop

mattdrepo, 2021-12-21:
couple of interesting pointers before i feedback your required info:

startx and loading i3 from my user, opening a terminal and typing libinput -listdevices allowed me for a brief instant to right click before hosing the system (no response from the keyboard or mouse before only the power down button allowed a normal shutdown).

libinput list-devices , after giving me the event number to try evemu-record, spewed some iichid info:

iichid0: iichid device close
iichid0: HID command I2C_HID_CMD_SET_POWER(1)
iichid0: tore callout down

evemu-record spews some numbers to the screen when moving my finger around - which is positive.

dmesg next message

mattdrepo, 2021-12-21:
ig4iic0: <Designware I2C Controller> iomem 0x91323000-0x91323fff irq 36 on acpi0
iicbus0: <Philips I2C bus (ACPI-hinted)> on ig4iic0
iicbus0: <unknown card> at addr 0x1a
ig4iic1: <Designware I2C Controller> iomem 0x91325000-0x91325fff irq 37 on acpi0
iicbus1: <Philips I2C bus (ACPI-hinted)> on ig4iic1
iichid_get_config_reg: error evaluating _DSM
iichid0:   IICbus addr       : 0x15
iichid0:   HID descriptor reg: 0x01
iichid0: HID command I2C_HID_CMD_DESCR at 0x1
iichid0: HID descriptor: 1e 00 00 01 74 00 02 00 03 00 22 00 04 00 00 00 05 00 06 00 f3 04 00 04 01 00 00 00 00 00
iichid0: <ELAN0000:03 04F3:0400 I2C HID device> at addr 0x15 on iicbus1
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: HID command I2C_HID_CMD_RESET
iichid0: Interrupt setup failed. Fallback to sampling
hidbus0: <HID bus> on iichid0
iichid0: HID command I2C_HID_REPORT_DESCR at 0x2 with size 116
iichid0: HID report descriptor: 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 75 01 95 02 81 02 95 06 81 03 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 05 0c 0a 38 02 95 01 81 06 75 08 95 03 81 03 c0 06 00 ff 85 0d 09 c5 15 00 26 ff 00 75 08 95 04 b1 02 85 0c 09 c6 96 76 02 75 08 b1 02 85 0b 09 c7 95 42 75 08 b1 02 09 01 85 5d 95 1f 75 08 81 06 c0
iichid0: HID command I2C_HID_CMD_SET_POWER(1)

mattdrepo, 2021-12-21:
more info with other ELAN devices that can have touchscreen and touchpad (in linux admittedly) going on about blacklisting ELAN0001 for ELAN0000 and vice versa:

https://askubuntu.com/questions/1248176/ideapad-5-15are05-elan-touchpad-not-working-on-20-04-nor-on-18-04

mattdrepo, 2021-12-21:
get a lot of ichid0: ff ff - ff when hitting the power to exit from the X session to power off.  Just to see what's attempting to go on behind the scenes with Xorg

mattdrepo, 2021-12-22:
Do you think this touchpad has a chance of working?

wulf7, 2021-12-22:
If evemu-record prints something like
```
E: 0.648029 0002 0000 -024	# EV_REL / REL_X                -24
E: 0.648029 0002 0001 -008	# EV_REL / REL_Y                -8
E: 0.648029 0000 0000 0001	# ------------ SYN_REPORT (1) ---------- +8ms
E: 0.656029 0002 0000 -035	# EV_REL / REL_X                -35
E: 0.656029 0002 0001 -014	# EV_REL / REL_Y                -14
E: 0.656029 0000 0000 0001	# ------------ SYN_REPORT (1) ---------- +8ms
```
every time you touch surface and something like
```
E: 0.000001 0001 0110 0001	# EV_KEY / BTN_LEFT             1
E: 0.000001 0000 0000 0001	# ------------ SYN_REPORT (1) ---------- +0ms
E: 0.160064 0001 0110 0000	# EV_KEY / BTN_LEFT             0
E: 0.160064 0000 0000 0001	# ------------ SYN_REPORT (1) ---------- +160ms
E: 0.680061 0001 0111 0001	# EV_KEY / BTN_RIGHT            1
E: 0.680061 0000 0000 0001	# ------------ SYN_REPORT (1) ---------- +192ms
E: 0.832064 0001 0111 0000	# EV_KEY / BTN_RIGHT            0
E: 0.832064 0000 0000 0001	# ------------ SYN_REPORT (1) ---------- +152ms
```
every time you press a button, than that means that touchpad is just work and all you did is break xorg config somewhere. Just remove you xorg.conf and start again.
If not, I need evemu-record output produced in system console rather than in X terminal with iichid debug enabled. It will cause interleaved evemu-record/iichid messages.

mattdrepo, 2021-12-22:
If not, I need evemu-record output produced in system console rather than in X terminal with iichid debug enabled. It will cause interleaved evemu-record/iichid messages.

Is their a way to output evemu-record to a txt file ?  I haven't tested eevemu-record any other way than console. And I didnt see results like those shown, just a spew of numbers as I moved and clicked.  Didn't configure Xorg - only set .xinitrc to exec i3

wulf7, 2021-12-22:
I want to see these numbers. Something like this should appear on system console after you get evemu-record started and surface touched:
```
iichid0: iichid device open
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: successfully setup callout
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 24 00 - 01 01 00 0f 05 5b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5a 00 01
iichid0: 24 00 - 01 01 00 0f 05 5b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b4 00 01
iichid0: 24 00 - 01 01 00 0f 05 5b 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 01 01
iichid0: 24 00 - 01 01 00 fc 04 78 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 68 01 01
iichid0: 24 00 - 01 01 00 ee 04 90 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c2 01 01
iichid0: 24 00 - 01 01 00 e5 04 a5 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1c 02 01
iichid0: 24 00 - 01 01 00 de 04 ba 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 76 02 01
iichid0: 24 00 - 01 01 00 dd 04 cf 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d0 02 01
iichid0: 24 00 - 01 01 00 e0 04 e2 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2a 03 01
iichid0: 24 00 - 01 01 00 f0 04 fd 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 84 03 01
iichid0: 24 00 - 01 01 00 04 05 0e 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 de 03 01
iichid0: 24 00 - 01 01 00 21 05 1e 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38 04 01
iichid0: 24 00 - 01 01 00 45 05 26 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 92 04 01
iichid0: 24 00 - 01 01 00 6c 05 28 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ec 04 01
iichid0: 24 00 - 01 01 00 93 05 23 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 46 05 01
iichid0: 24 00 - 01 01 00 cd 05 16 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a0 05 01
iichid0: 24 00 - 01 01 00 ef 05 0c 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa 05 01
iichid0: 24 00 - 01 01 00 0b 06 03 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 06 01
iichid0: 24 00 - 01 01 00 24 06 f8 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ae 06 01
iichid0: 24 00 - 01 01 00 35 06 ee 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 08 07 01
iichid0: 24 00 - 01 01 00 43 06 e7 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 62 07 01
iichid0: 24 00 - 01 01 00 4c 06 e3 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 bc 07 01
iichid0: 24 00 - 01 01 00 51 06 e5 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 16 08 01
iichid0: 24 00 - 01 01 00 52 06 f4 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 70 08 01
iichid0: 24 00 - 01 01 00 51 06 08 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ca 08 01
iichid0: 24 00 - 01 01 00 51 06 1f 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 24 09 01
iichid0: 24 00 - 01 01 00 56 06 36 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7e 09 01
iichid0: 24 00 - 01 01 00 5b 06 4f 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d8 09 01
iichid0: 24 00 - 01 01 00 5b 06 70 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 0a 01
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: 00 00 - 00
iichid0: iichid device close
iichid0: 00 00 - 00
iichid0: HID command I2C_HID_CMD_SET_POWER(1)
iichid0: tore callout down
```
You can copy them from dmesg ouput later.

mattdrepo, 2021-12-23:
hope this helps!

pcib1: <PCI-PCI bridge> at device 28.0 on pci0
pcib2: <ACPI PCI-PCI bridge> at device 28.2 on pci0
pci1: <ACPI PCI bus> on pcib2
pci1: <network> at device 0.0 (no driver attached)
isab0: <PCI-ISA bridge> at device 31.0 on pci0
isa0: <ISA bus> on isab0
atkbdc0: <Keyboard controller (i8042)> port 0x60,0x64 on acpi0
atkbd0: <AT Keyboard> irq 1 on atkbdc0
kbd0 at atkbd0
atkbd0: [GIANT-LOCKED]
ig4iic0: <Designware I2C Controller> iomem 0x91323000-0x91323fff irq 36 on acpi0
iicbus0: <Philips I2C bus (ACPI-hinted)> on ig4iic0
iicbus0: <unknown card> at addr 0x1a
ig4iic1: <Designware I2C Controller> iomem 0x91325000-0x91325fff irq 37 on acpi0
iicbus1: <Philips I2C bus (ACPI-hinted)> on ig4iic1
iichid_get_config_reg: error evaluating _DSM
iichid0:   IICbus addr       : 0x15
iichid0:   HID descriptor reg: 0x01
iichid0: HID command I2C_HID_CMD_DESCR at 0x1
iichid0: HID descriptor: 1e 00 00 01 74 00 02 00 03 00 22 00 04 00 00 00 05 00 06 00 f3 04 00 04 01 00 00 00 00 00
iichid0: <ELAN0000:03 04F3:0400 I2C HID device> at addr 0x15 on iicbus1
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: HID command I2C_HID_CMD_RESET
iichid0: Interrupt setup failed. Fallback to sampling
hidbus0: <HID bus> on iichid0
iichid0: HID command I2C_HID_REPORT_DESCR at 0x2 with size 116
iichid0: HID report descriptor: 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 75 01 95 02 81 02 95 06 81 03 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 05 0c 0a 38 02 95 01 81 06 75 08 95 03 81 03 c0 06 00 ff 85 0d 09 c5 15 00 26 ff 00 75 08 95 04 b1 02 85 0c 09 c6 96 76 02 75 08 b1 02 85 0b 09 c7 95 42 75 08 b1 02 09 01 85 5d 95 1f 75 08 81 06 c0
iichid0: HID command I2C_HID_CMD_SET_POWER(1)

iichid0: iichid device open
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: successfully setup callout
iichid0: ff 00 - ff
iichid0: iichid device close
iichid0: HID command I2C_HID_CMD_SET_POWER(1)
iichid0: tore callout down
iichid0: iichid device open
iichid0: HID command I2C_HID_CMD_SET_POWER(0)
iichid0: successfully setup callout
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: 22 00 - 5d 08 51 8d f0 33 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 f3
iichid0: 22 00 - 5d 08 51 88 f2 33 0f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0c 00 00 00 66
iichid0: 22 00 - 5d 08 51 84 f5 33 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 5c
iichid0: 22 00 - 5d 08 52 7e 2f 43 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 4c
iichid0: 22 00 - 5d 08 52 8a b7 33 15 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 00 00 5e
iichid0: 22 00 - 5d 08 53 cb 27 33 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 00 00 24
iichid0: 22 00 - 5d 08 63 16 5e 33 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 45
iichid0: 22 00 - 5d 08 63 52 85 33 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 00 00 a1
iichid0: 22 00 - 5d 08 63 84 b1 32 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 00 00 6f
iichid0: 22 00 - 5d 08 63 a6 ce 33 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 00 00 35
iichid0: 22 00 - 5d 08 63 dd da 33 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 ce
iichid0: 22 00 - 5d 08 73 23 e2 34 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 f1
iichid0: 22 00 - 5d 08 73 67 cc 33 15 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0f 00 00 00 c2
iichid0: 22 00 - 5d 08 73 b1 ab 33 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 3e
iichid0: 22 00 - 5d 08 73 e0 7e 32 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 50
iichid0: 22 00 - 5d 08 83 0b 58 33 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 03
iichid0: 22 00 - 5d 08 83 2b 2b 33 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 f5
iichid0: 22 00 - 5d 08 83 33 01 43 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 18
iichid0: 22 00 - 5d 08 82 31 ea 33 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 3d
iichid0: 22 00 - 5d 08 82 2d dd 33 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 e4
[... 344 lines cut ...]
iichid0: 22 00 - 5d 09 62 bb 08 33 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 00 00 00 b9
iichid0: 22 00 - 5d 09 62 bb 08 33 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 13 00 00 00 1f
iichid0: 22 00 - 5d 09 62 bb 08 33 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 13 00 00 00 1f
iichid0: 22 00 - 5d 08 62 bb 08 33 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 13 00 00 00 d2
iichid0: 22 00 - 5d 08 62 bb 08 33 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 00 00 00 c4
iichid0: 22 00 - 5d 08 62 bb 08 33 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 00 00 00 c4
iichid0: 22 00 - 5d 08 62 bb 08 33 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 00 4e
iichid0: 22 00 - 5d 08 62 bb 08 33 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0d 00 00 00 b6
[... cut ...]
iichid0: 22 00 - 5d 08 62 bb 08 11 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 da
iichid0: 22 00 - 5d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00 50
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: ff ff - ff
iichid0: iichid device close
iichid0: ff ff - ff
iichid0: HID command I2C_HID_CMD_SET_POWER(1)
iichid0: tore callout down

mattdrepo, 2021-12-23:
that was me basically, scrolling, clicking, two finger clicking etc with pauses.

mattdrepo, 2021-12-24:
when i exit i3, i get a lot of  iichid0: ff ff - ff - debug messages.  It does feel like something is blocking the i2c to this machine.  Sometimes I am able to register a right click (this happened once when i was certain I wasn't using my USB mouse and I confirmed it when opening an app that might of allowed the i2c access...?).

wulf7, 2021-12-24:
It seems that the device starts in native absolute mode rather than in mouse emulation. FreeBSD does not support it yet. I ported support for it from Linux some time ago: https://github.com/wulf7/iichid/blob/master/hetp.c but it has never been committed to FreeBSD due to lack of testing. If you are willing to test it, I can do a port to 14-CURRENT/13-STABLE.

mattdrepo, 2021-12-24:
yes please!!

mattdrepo, 2021-12-24:
can you explain in steps i can follow?

mattdrepo, 2021-12-24:
So I don't end up with something unnecessary, could you confirm whether I would need to do the earlier patches you told me about.  And would this mean doing a git pull of your source.  I am a bit confused as to how to combine, or whether to just get the freebsd kernel source and add your file somehow.  Thank you and Merry Christmas!

mattdrepo, 2021-12-24:
Ah yes I see what you mean at present : in that it wont "make" on FreeBSD 13 STABLE

mattdrepo, 2021-12-29:
do you think it will be possible to make a patch for me to try on my system?

wulf7, 2021-12-29:
I going to make a patch after getting some spare time at the beginning of new year

wulf7, 2022-01-06:
I created driver for 14-CURRENT called ietp. It should be applicable to 13-STABLE too. Patch is located here: https://people.freebsd.org/~wulf/ietp.patch
It is not tested though.

mattdrepo, 2022-01-06:
Couple of quick questions:

Do i do the same things as before regarding rc.conf and could you explain how to use this patch file (i previously manually applied the edits).  Thanks again and happy new year.

wulf7, 2022-01-07:
> Do i do the same things as before regarding rc.conf

Yes, keep all the changes you did to `/boot/loader.conf` and kernel config, than take vanilla system sources and do
```
# cd /usr/src
# patch -p1 < /path/to/ietp.patch
```
than check that it is applied cleanly and rebuild kernel.
Than reboot and post text of kernel panic which is most probably will happen during a boot. I am sure that there are bugs in driver :-)
You may apply very first patch from this thread to get ig4.ko loaded automatically on boot, but alternatively you can load it with any other method.

Happy new year!

wulf7, 2022-01-07:
I made small bugfix and updated patch on old place.
If you have applied old patch already, than you can just apply this small patch on top of old one.
```
--- sys/dev/hid/ietp.c
+++ sys/dev/hid/ietp.c
@@ -142,7 +142,7 @@ static int		ietp_iic_write_reg(device_t, uint16_t, uint16_t);
 static int		ietp_iic_set_absolute_mode(device_t, bool);
 
 #define	IETP_IIC_DEV(pnp) \
-	{ HID_TLC(0xff00, 0x0001), HID_BUS(BUS_I2C), HID_PNP(pnp) }
+    { HID_TLC(HUP_GENERIC_DESKTOP, HUG_MOUSE), HID_BUS(BUS_I2C), HID_PNP(pnp) }
 
 static const struct hid_device_id ietp_iic_devs[] = {
 	IETP_IIC_DEV("ELAN0000"),
```

mattdrepo, 2022-01-07:
no I haven't as of yet.  I will have to back up this machine again and have another go.  Will update shortly as to the progress.

Is the link to the patch now including this bugfix?

wulf7, 2022-01-07:
Yes, patch under the link includes bugfix now.

mattdrepo, 2022-01-10:
Hi there.

On enabling your patch I cannot log in after a reboot.  Only the enter key works

mattdrepo, 2022-01-10:
hmm might be my error by the looks of it - got an unterminated quoted string error on the following reboot - must of made a mistake.  I cannot log in.

mattdrepo, 2022-01-10:
how do i make my filesystem writeable when making an error on rc.conf and having to log in as root?  (seems i forgot to encapsulate an entry, got left with a broken log in, had to log in as root and realised my keyboard was defaulting to US)

mattdrepo, 2022-01-10:
zfs set readonly=off zroot

just installing a window manager to see what happens when i get to a desktop

[44 more replies]

closed 2022-02-14 by wulf7