wulf7/iichid #64: Asus C300 report length too big, trackpad only works for a little bit when ignored

obiwac, 2023-01-03:
Hey!

I've been trying to debug this for a little while and I admit defeat; my Elantech touchpad (`_HID=ELAN0000`) doesn't work under what is essentially 14-CURRENT on my Asus C300 Chromebook.

Compiling with `IICHID_DEBUG` and setting `hw.iichid.debug=5`, I'm getting a lot of `input report too big. requested=28 received=32` messages in dmesg. I'll try to read into this a little more tomorrow, but so far this seems odd to me, as `IETP_REPORT_LEN_LO` is 32, so 32 bytes should be requested, not 28.

I added a `len += 4` in `ietp_intr` to see if I could go any further, and it seems like it works fine-ish for a little bit after running `libinput debug-events`, before crapping itself, giving garbage data, and, eventually, straight up dying.

Where could I go from here wrt debugging?

Wishing you a happy new year 🍾🥂

obiwac, 2023-01-03:
Small update: I've gotten the trackpad to work perfectly well (no stuttering, works continuously) by forcibly changing the value of `isize` in the HID report descriptor to `32` (in `hidbus_fill_rdesc_info`).

Now I'm trying to figure out why `hid_report_size_max` is returning `28` instead of `32`. If I do, I'll submit a fix directly to FreeBSD - I imagine there isn't any use opening a PR on this repo :)

wulf7, 2023-01-08:
I think HID report descriptor is broken. ELAN0000 is not a HID device in native mode so report descriptor is not used at all. But our HID subsystem still calculates input report size based on it.
Increase of isize is a proper solution and can be done via overloading of report descriptor in `device_identify` handler. See e.g. `bcm5974_identify` subroutine in apple touchpad driver `sys/dev/hid/bcm5974.c`. In ELAN0000 we should do an extra step and compare `hid_report_size_max` result with 32. And than load dummy descriptor if device's one is undersized.

obiwac, 2023-01-09:
Thank you. I'll look at implementing a proper fix after my exams ;)

obiwac, 2023-02-03:
> ELAN0000 is not a HID device in native mode so report descriptor is not used at all

Are you sure about this? Because `hid_get_report_descr()` seems to still return what looks like a valid HID report descriptor. Maybe I misunderstand what you mean by "native mode". The report descriptor does look a little bit mangled in places though - do you think there's a possibility FreeBSD reads them incorrectly? I'll look at how Linux does it when I have time.

Locally I'm now overloading `device_indentify` and loading in a dummy report descriptor if `hid_report_size_max()` returns something smaller than `IETP_REPORT_LEN_LO`, similar to the Apple touchpad driver, as you suggested:

```c
static const uint8_t ietp_dummy_rdesc[] = {
	0x05, HUP_GENERIC_DESKTOP,	/* Usage Page (Generic Desktop Ctrls)	*/
	0x09, HUG_MOUSE,		/* Usage (Mouse)			*/
	0xA1, 0x01,			/* Collection (Application)		*/
	0x09, 0x01,			/*   Usage (0x01)			*/
	0x95, IETP_REPORT_LEN_LO,	/*   Report Count (IETP_REPORT_LEN_LO)	*/
	0x75, 0x08,			/*   Report Size (8)			*/
	0x81, 0x02,			/*   Input (Data,Var,Abs)		*/
	0xC0,				/* End Collection			*/
};
```

I'll submit a patch shortly :)

wulf7, 2023-02-05:
> Are you sure about this?

Absolutely. Trackpad has a valid HID report descriptor for compatibility mode when it emulates standard HID relative mouse. But after switching to native mode it becomes non-HID device. Just look at Linux driver: https://github.com/torvalds/linux/blob/master/drivers/input/mouse/elan_i2c_core.c
The main reason ietp(4) is built on base of HID rather then I2C bus is missing GPIO interrupt support. With using HID we can reuse I2C polling code from iichid driver.

obiwac, 2023-02-05:
Right, I had a misunderstanding about what native mode referred to.

closed 2023-02-05 by obiwac